libretiny-eu/ltchiptool #5: Flashing Tuya WR2

Sonic-Amiga, 2023-06-27:


kuba2k2, 2023-06-27:
Realtek has a total of three UART ports (UART0, UART1, UART2). Make sure you're using the UART2's TXD, not any other. It might be not exposed on a pin, but a test point for instance (and that seems to be the case on WR2).

As for the weird CEN behavior, I'm not sure. CEN should be a forced power-down of the chip. You can also apply power while pulling TXD down, it will work as well. But it has to be TX2, not TX0.

Sonic-Amiga, 2023-06-28:
Thank you for your quick response; i will test it later. Examined the board closely; perhaps i see them 
 Interesting that datasheet doesn't mention them in any way; it only shows UART0.
 Ve-e-e-e-ry unfriendly hardware 👎

kuba2k2, 2023-06-28:
Take a look at this photo I found [on FCC](https://fccid.io/2ANDL-WR2/Internal-Photos/Int-Photos-4045815.pdf):
[image]

A29 and A30 are UART2_RX and UART2_TX, respectively. If you confirm that there actually is a serial port on these pins, I'll add this to [the pinout diagram in LT docs](https://docs.libretiny.eu/boards/wr2/#pinout), so that it's no longer a mystery.

Sonic-Amiga, 2023-06-28:
I was also looking at the same two pads, because there's simply nothing else. Will test on Saturday i think.
 I guess this deserves the highest difficulty score: full device disassembly required.

Sonic-Amiga, 2023-07-01:
Hello! I'm back, just succeeded with getting debug log from my WR02. Unfortunately i forgot to take a pic of my removed module before soldering wires; so here is what i've got. Note slightly different placement of connection points; one on top of another.
[image]
It's also possible to attach ground to the ring shown; this will allow to re-assemble the device while having convenient serial port access.

Sonic-Amiga, 2023-07-01:
Here's a high-res photo of an assembled board of my smart plug; with WR02 sitting on it. Debug pins are practically inaccessible
[image]

Sonic-Amiga, 2023-07-01:
So far, i've put together a little adapter board with "flash mode" button for my convenience
[image]
The button simply shorts TX line (WR2 side) to ground. I think i can now enter flash mode. After powering up the module while the button is pressed, i now see a "garbage character" coming in, 3-4 per second. Analysis with RealTerm shows that it reads as 0xFF, regardless of port speed setting.

My next problem: i try to read the stock firmware as a test; and ltchiptool can't connect. It simply sits and cycles between "connecting... soft reset... connecting... hard reset...". I tried leaving the speed at "auto" as well as setting it to 115200. The chip i select is "Realtek amebaZ".

The serial comm works 100% If i just plug the module in with no button pressed, i am able to read the log coming from the device at 115200 8n1.

Sonic-Amiga, 2023-07-01:
BTW, if i type something in the terminal while the module is booted up, it echoes characters back. So comms 200% working.

Sonic-Amiga, 2023-07-01:
Tried setting ltchiptool's log level to 'verbose', it doesn't say anything really interesting:
```
D: Connecting
V: OnUpdate(busy=set())
V: OnUpdate(busy={'com3'})
D: Connecting
V: OnUpdate(busy=set())
V: OnUpdate(busy={'com3'})
D: Connecting
V: OnUpdate(busy=set())
V: OnUpdate(busy={'com3'})
D: Connecting
V: OnUpdate(busy=set())
V: OnUpdate(busy={'com3'})
```
and that's it

Sonic-Amiga, 2023-07-01:
Looked into ltchiptool source code for possibly more explanations. First of all, it appears the chip should actually send 0x00, not 0xFF. Second, what a strange baud rate:
RTL_ROM_BAUD = 1500000
I tried setting it in RealTerm by hands; and i've got an error: "this baud rate isn't supported by your adapter". Strangely, ltchiptool doesn't say anything at all when i try to select it for the port in terminal mode. It stops receiving the log, however.

Is it known, what USB-To-Serial adapters are supported ? I am using PL2303

Sonic-Amiga, 2023-07-01:
Played for some more; tried official RTL image tool. No result. Since it can only erase and flash images; i tried to issue "read flash status register" command using "..." button. It times out with no response.

Also i noticed that in the terminal, while the chip is sending its "garbage character", i type something, it stops sending it, but doesn't do anything else after that. I have to powercycle.

And, just for complete info, here is my module's boot log
```
ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1168:10002000]
IMG1 ENTRY[8000541:100021ef]
IMG1 ENTER
CHIPID[000000ff]
read_mode idx:0, flash_speed idx:0
calibration_result:[1:9:9][5:d]
calibration_result:[2:13:7][1:d]
calibration_result:[3:3:3][1:3]
calibration_ok:[2:13:7]
FLASH CALIB[NEW OK]
OTA2 ADDR[80d0000]
OTAx SELE[fffffffc]
OTA1 USE
IMG2 DATA[0x809fb00:6564:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x800b1a5]
===== Enter Image 2 ====
System_Init1
OSC8M: 8390004
boot reason: 0
ROM:[V0.1]
ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1168:10002000]
IMG1 ENTRY[8000541:100021ef]
IMG1 ENTER
CHIPID[000000ff]
read_mode idx:0, flash_speed idx:0
calibration_result:[1:9:9][5:d]
calibration_result:[2:13:7][1:d]
calibration_result:[3:3:3][1:3]
calibration_ok:[2:13:7]
FLASH CALIB[NEW OK]
OTA2 ADDR[80d0000]
OTAx SELE[fffffffc]
OTA1 USE
IMG2 DATA[0x809fb00:6564:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x800b1a5]
===== Enter Image 2 ====
System_Init1
OSC8M: 8386568
boot reason: 0
System_Init2

11111111111111111111111111
interface 0 is initializ[01-ed
interfa01 18:12:15 ce 1 is iniTUYA Notice]tialized

Initializie.c:244] < Tng WIFI ...UYA IOT SDK
LDO MV:2.0.0 BS:3ode, BD_Inf0.04_PT:2.2_o: 0
.0 >3.3_CAD:1.0.2_CD:1.0
< tuy
LDO Mode,a_iot_lib BU BD_Info: 0ILD AT:2019_
06_21_14_56_08 BY tuya_iot_team AT 8710_2M >
IOT DEFS < WIFI_GW:1 DEBUG:1 KV_FILE:0 SHUTDOWN_MODE:0 LITTLE_END:1 TLS_MODE:3 ENABLE_LOCAL_LINKAGE:0 ENABLE_CLOUD_OPERATION:0 ENABLE_SUBDEVICE:0 ENABLE_ENGINEER_TO_NORMAL:0 OPERATING_SYSTEM:2 ENABLE_SYS_RPC:0 TY_SECURITY_CHIP:0 RELIABLE_TRANSFER:RELIABLE_TRANSFER ENABLE_LAN_ENCRYPTION:1 ENABLE_SIGMESH:0 >

[01-01 18:12:15 TUYA Notice][tuya_device.c:245] oem_rtlbn_dltj_2M:1.0.4
[01-01 18:12:15 TUYA Notice][tuya_device.c:246] firmware compiled at Jul 23 2019 17:03:22
[01-01 18:12:15 TUYA Notice][simple_flash.c:428] key_addr: 0x1eb000   block_sz 4096
[01-01 18:12:15 TUYA Notice][simple_flash.c:496] get key:
0xf3 0x9e 0x78 0xc4 0x95 0x96 0xb1 0x5a 0xa7 0x32 0x97 0x3 0x76 0xec 0x96 0xa7

WIFI initialized

init_thread(55), Available heap 0x9ea0[01-01 18:12:15 TUYA Notice][tuya_uart.c:125] 1   9600
[01-01 18:12:15 TUYA Notice][tuya_main.c:368] mf_init succ
[01-01 18:12:15 TUYA Notice][json_config.c:115] ===channel[0] exist===
[01-01 18:12:15 TUYA Notice][json_config.c:140] key:  pin=18, drive level=low
[01-01 18:12:15 TUYA Notice][json_config.c:141] relay:  pin=0, drive level=high
[01-01 18:12:15 TUYA Notice][json_config.c:142] led : not exist
[01-01 18:12:15 TUYA Notice][json_config.c:146] init_stat: CLOSE
[01-01 18:12:15 TUYA Notice][json_config.c:149] dpid:1, cd_dpid:9
[01-01 18:12:15 TUYA Notice][json_config.c:162] total key: not exist
[01-01 18:12:15 TUYA Notice][json_config.c:163] total led: not exist
[01-01 18:12:15 TUYA Notice][json_config.c:241] wifi led:  pin=23, drive level=low
[01-01 18:12:15 TUYA Notice][json_config.c:244] when wifi unconnect,wifi led stat:ON
[01-01 18:12:15 TUYA Notice][json_config.c:247] when wifi connect,wifi led stat:DIR RELAY
[01-01 18:12:15 TUYA Notice][json_config.c:263] work mode: LOW_POWER
[01-01 18:12:15 TUYA Notice][json_config.c:269] button reset time: 3000 mS
[01-01 18:12:15 TUYA Notice][json_config.c:383] get chip type:bl0937
[01-01 18:12:15 TUYA Notice][json_config.c:484] epin = 5, ivpin = 12, ivcpin = 15, ivcpin_type = HIGH, v_ref = 586, i_ref = 28928, p_ref = 16929, e_ref= 1408, v_def = 2200, i_def = 392, p_def = 864
[01-01 18:12:15 TUYA Notice][json_config.c:429] e_dp:17, i_dp:18, p_dp:19, v_dp:20, fault_dp:26
[01-01 18:12:15 TUYA Notice][json_config.c:430] ri: 1, over_current: 17000

LwIP_DHCP: dhcp stop.
Deinitializing WIFI ...
WIFI deinitialized
Initializing WIFI ...
LDO Mode, BD_Info: 0

LDO Mode, BD_Info: 0

WIFI initialized
[01-01 18:12:16 TUYA Notice][tuya_main.c:116] current product ssid name:tuya_mdev_test2
[01-01 18:12:17 TUYA Notice][wifi_hwl.c:268] not find scan ssid (tuya_mdev_test2)
[01-01 18:12:17 TUYA Notice][tuya_main.c:120] wf_assign_ap_scan failed(-606)
[01-01 18:12:17 TUYA Notice][gw_intf.c:2600] serial_no:708976dc312d
[01-01 18:12:17 TUYA Notice][gw_intf.c:2631] gw_cntl.gw_wsm.stat:1
[01-01 18:12:17 TUYA Notice][gw_intf.c:2634] gw_cntl.gw_wsm.nc_tp:0
[01-01 18:12:17 TUYA Notice][gw_intf.c:2635] gw_cntl.gw_wsm.md:0
[01-01 18:12:17 TUYA Notice][gw_intf.c:2667] gw_cntl.gw_if.abi:0 input:0
[01-01 18:12:17 TUYA Notice][gw_intf.c:2668] gw_cntl.gw_if.product_key:keydnqmh87c8ajv4, input:keydnqmh87c8ajv4
[01-01 18:12:17 TUYA Notice][gw_intf.c:2669] gw_cntl.gw_if.tp:0, input:0
[01-01 18:12:17 TUYA Notice][gw_intf.c:2671] gw_cntl.gw_if.firmware_key:keydnqmh87c8ajv4, input:keydnqmh87c8ajv4
[01-01 18:12:17 TUYA Notice][tuya_gpio.c:225] id 18 {0x100182dc}
[01-01 18:12:17 TUYA Notice][app_dltj.c:227] have not storage ele data:-901
[01-01 18:12:17 TUYA Notice][app_dltj.c:231] have not storage tem ele data:-901
[01-01 18:12:17 TUYA Notice][tuya_device.c:293] device_init ok ! free_mem_size:38576

LwIP_DHCP: dhcp stop.
Deinitializing WIFI ...
WIFI deinitialized
[rltk_wlan_rf_off] netif is DOWN
WIFI is not running
[rltk_wlan_rf_off] netif is DOWN[01-01 18:12:17 TUYA Err][gw_intf.c:360] wf_wk_mode_set err:-1
[01-01 18:12:18 TUYA Notice][app_switch.c:1222] wifi status is :0
```

kuba2k2, 2023-07-01:
It's possible that PL2303 doesn't support that baudrate. 1500000 is the baudrate used during linking phase, so it has to be supported. Maybe ltchiptool doesn't know that your adapter can't support it, and realterm somehow does.

Typing characters should not echo them back, unless local echo is enabled in your terminal program. The chip should be sending 0x15 on 1500000 baudrate, not 0x00 nor 0xFF. Try with a different adapter maybe.

Sonic-Amiga, 2023-07-02:
Confirmed. Went to a shop today and bought an FT232 adapter. Worked like a charm! Original image successfully downloaded.
Interestingly, after i set a proper baud rate in RealTerm, i see a chip only sending me a single 0x00 when entering flash mode.

Nevertheless, thank you for your quick replies. Feel free to use what we've found out here, with photos, for a documentation.

kuba2k2, 2023-07-02:
I'll transfer this one to the main LibreTiny repo, and keep it open so that I remember to add the missing pins to the docs.

Sonic-Amiga, 2023-07-02:
BTW, do you have device .yml repo somewhere, like the original project has ?

kuba2k2, 2023-07-02:
We only have a Discord channel for that. You can find a link in the docs main page.

Sonic-Amiga, 2023-07-02:
Okay, i'll have a look

kuba2k2, 2023-07-13:
Fixed in https://github.com/kuba2k2/libretiny/pull/147.

closed 2023-07-13 by kuba2k2