libretiny-eu/ltchiptool #7: Error when creating uf2 file from OpenBK firmware

MnM001, 2023-08-02:
Hi,

I have a device that I will like to flash with OpenBK firmware.
On my Ubuntu I have downloaded ltchiptool and did:

 `python3 -m ltchiptool uf2 write -b cb2s -o openbkt.uf2 OpenBK7231N_1.17.205.rbl=device:download`

and a few other variants of that command. I always get this error:

`Error: Invalid value for '-b' / '--board': Board cb2s does not exist`

I am not sure what I am doing wrong. Can you please help?

kuba2k2, 2023-08-02:
You need to have LibreTiny installed for this so that it finds the board IDs. It usually comes with ltchiptool when you install from pip, but if you install from source instead you need LT as well.

MnM001, 2023-08-02:
I am installing ltchiptool  wiht pip:

`sudo pip install ltchiptool`

However I always get that error.

I have installed LibreTiny too and I think it works OK:

` % sudo python3 -m esphome
usage: __main__.py [-h] [-v] [-q] [-s key value] command ...
__main__.py: error: the following arguments are required: command
`
and

` % sudo python3 -m esphome wizard yourdevice.yml
Hi there!
I'm the wizard of ESPHome :)
And I'm here to help you get started with ESPHome.
In 4 steps I'm going to guide you through creating a basic configuration file for your custom ESP8266/ESP32 firmware. Yay!

============= STEP 1 =============
    _____ ____  _____  ______
   / ____/ __ \|  __ \|  ____|
  | |   | |  | | |__) | |__
  | |   | |  | |  _  /|  __|
  | |___| |__| | | \ \| |____
   \_____\____/|_|  \_\______|

===================================
First up, please choose a name for your node.
It should be a unique name that can be used to identify the device later.
For example, I like calling the node in my living room livingroom.
`

But this is where all stops. Running the python3 -m ltchiptool command only gets me  the same error (regardless whats after -b (including bk7231n or bk7231t):

`Error: Invalid value for '-b' / '--board': Board cb2s does not exist`

I don't know what I am doing wrong so your assistance is still required.

kuba2k2, 2023-08-02:
Try `ltchiptool -vv list boards` and post the output.

MnM001, 2023-08-02:
Here it is 

`+------+------+-------------------+-------------+
| Name | Code | MCU / Flash / RAM | Family name |
+------+------+-------------------+-------------+
+------+------+-------------------+-------------+
(END)
`

kuba2k2, 2023-08-03:
Well, you're installing ltchiptool as sudo but running it without sudo.

Uninstall all versions of ltchiptool (`pip uninstall ltchiptool` and `sudo pip uninstall ltchiptool`), then install it *without* sudo and run the boards command again.

It should look like this, if it finds the LibreTiny directory:
```
~  λ ltchiptool -vv list boards
D: Found PlatformIO Package (v1.2.0) - C:\Users\Kuba\.platformio\platforms\libretiny
D: Found Local data snapshot (v1.0.2) - D:\Dev\tuya\tool-ltchiptool\ltchiptool
+--------------------------------+----------------------------+-----------------------------+---------------+
| Name                           | Code                       |      MCU / Flash / RAM      |  Family name  |
+--------------------------------+----------------------------+-----------------------------+---------------+
| BW12                           | bw12                       | RTL8710BX / 2 MiB / 256 KiB |  realtek-ambz |
| BW15                           | bw15                       | RTL8720CF / 2 MiB / 256 KiB | realtek-ambz2 |
.....
```

MnM001, 2023-08-03:
I don't believe it was that simple :)
Thank you for helping me with his - I now get a list of boards.

I will try and covert the firmware later. Will reply here with the result.

MnM001, 2023-08-03:
This the command I used that worked for me (for the T version)

`'/home/XXXX/.local/bin/ltchiptool' uf2 write -b generic-bk7231n-qfn32-tuya -o /tmp/OpenBK7231N_1.17.205.uf2 'OpenBK7231N_1.17.205.rbl=device:download'`

tellblom, 2023-08-21:
Tried the above and installed using pip install ltchiptool  ( not using sudo) 
I can run without sudo ( if I try to use sudo with it won't run)

I still get the error:  Error: Invalid value for '-b' / '--board': Board generic-bk7231n-qfn32-tuya does not exist

Get no boards just 
D: Found Local data snapshot (v1.1.0) - /home/xxx/.local/lib/python3.9/site-packages/ltchiptool

+------+------+-------------------+-------------+
| Name | Code | MCU / Flash / RAM | Family name |
+------+------+-------------------+-------------+
+------+------+-------------------+-------------+

Any ideas?

BenJamesAndo, 2023-08-31:
Is there any easy way to help ltchiptool know where the libretiny package is? I did however manag to get it to find it by extracting libretiny into the same folder as ltchiptool on my Windows PC and running `python3.11 -m ltchiptool uf2 write -b generic-bk7231t-qfn32-tuya -o OpenBeken.uf2 "OpenBK7231T_1.17.230.rbl=device:download"`

rmarquet, 2023-09-26:
I'm having the issue described above - I just installed ltchiptool with "`pip install ltchiptool`", and it installed it in my .local/ directory.  When I run the "`ltchiptool -vv list boards`" command, I get:

+------+------+-------------------+-------------+
| Name | Code | MCU / Flash / RAM | Family name |
+------+------+-------------------+-------------+
+------+------+-------------------+-------------+

But then after I hit Q to exit less, I get this message (username redacted):
`D: Found Local data snapshot (v1.3.0) - /home/<username>/.local/lib/python3.10/site-packages/ltchiptool`

Which strongly implies that it found it. The board.py, enums.py, and family.py files are in the models/ directory under that path.  Board.py, at least, has data in it; it doesn't appear to be corrupted.

I tried giving it a -L parameter but no path I try results in success, either.  (For what it's worth, I also tried it with sudo after uninstalling the normal user copy, and it didn't work then, either.  I removed that and went back to the user installed version, no improvement.)

Running kubuntu 22.04.  Debian version is bookworm-sid.  Thanks for any suggestions!

MnM001, 2023-09-27:
For anyone who is stuck and needs the files another user posted in the OpenBeken forums a link to the UF2 files:

https://github.com/BenJamesAndo/OpenBeken_uf2_firmware/releases/tag/1.17.230

rmarquet, 2023-09-28:
> For anyone who is stuck and needs the files another user posted in the OpenBeken forums a link to the UF2 files:
> 
> https://github.com/BenJamesAndo/OpenBeken_uf2_firmware/releases/tag/1.17.230

Thanks.  Turned out libretiny was better for my application after all!

Marcin685, 2026-01-14:
Hello, Could someone convert the current version of OpenLN882H_1.18.245 to UF2 format so I can flash it to Esphome over-the-air? 

[image]

The conversion instructions require an .rbl file, but OB doesn't provide .rbl files for the LN882H. I tried converting it using the command `ltchiptool uf2 write -b generic-ln882hki -o firmware.uf2 OpenLN882H_1.18.245.bin=device:download` , but the `ltchiptool uf2 info firmware.uf2` preview doesn't show the partition table.