libretiny-eu/ltchiptool #8: Microsoft Defender detects trojan in newest Release

panzerknacker20111, 2023-08-29:
Hello,

I wanted to try your newest release, but my Microsoft Defender immediatly raised an alarm and deleted the downloadet file.

It shows the following:

[image]

The previous Release download shows no problems at all.

Just wanted to make you aware of this. Maybe it is false positive but maybe not. I will always trust defender first ;)

sincze, 2023-09-02:
Same here indeed:
[image]

kuba2k2, 2023-09-02:
I tried to exclude various modules and revert the code, but sadly I got the same results.

The only advice for now is to either ignore the virus alerts (false positives), or install ltchiptool using Python via `pip install ltchiptool`.

panzerknacker20111, 2023-09-06:
I tested the new 4.4.1 release and defender had nothing to complain. great. Thank you for your work. This issue is resolved with that release I think.

closed 2023-09-06 by panzerknacker20111