libretiny-eu/ltchiptool #12: Problems flashing a new rtl8710bx board

protectivedad, 2023-10-26:
I'm working on a plug HPPA11SW that has a board in it with a rtl8710bx chip. I was able to use the ltchiptool to read the flash no problem. But writing the flash sends information and then errors "Failed to write to 0x80000000" or whatever address I am writing to. I ran it with `--dump-serial` the last lines are:
```
S: <- TX: FF FF FF FF FF FF FF 1A                           ........
S: -> RX: b'\x15'
E: ValueError: Failed to write to 0x8000000
E: |-- File "/usr/local/lib/python3.10/dist-packages/ltchiptool/soc/ambz/flash.py", line 148, in flash_write_raw
```

Note: I can write using the amebaz-image-tool-v2.2.

The board is an unsupported one, that I am trying to get working. If it is helpful it is shown in FCC ID filings 2AB2QHPPA11SZ and 2AB2QHPPA11SZ (it is the same internal board different plug) with markings RL-M, E350220. M-LWK32F510D. Later I was going to look for help adding it to libretiny, but first I wanted to get the ltchiptool working.

kuba2k2, 2023-10-26:
It is supported, if it has RTL8710BX chip.

How far along does it get before it shows that error? Does it terminate at the beginning or further in the process?

Are you sure the power supply is powerful enough for to power the board?

ferbulous, 2023-10-26:
I actually had the same problem, i just keep repeat flashing until it’s completed.

protectivedad, 2023-10-26:
> It is supported, if it has RTL8710BX chip.
>
I meant it is not list in the "boards" list. Plus it seems to use a 700k app area, which is causing me issues. But that is for a different discussion.

> How far along does it get before it shows that error? Does it terminate at the beginning or further in the process?
>

It seems like it is sending 51909 bytes.
> Are you sure the power supply is powerful enough for to power the board?

No, but I haven't looked into it because it works using the amebaz-image-tool-v2.2. If it was a power supply problem I figure that wouldn't work either.

Just a few points to clarify. I don't have the rts/dtr setup. I manually put the board into download mode. The command I run is:
```
ltchiptool flash write -d /dev/ttyUSB0 -f ambz hppa11sw-c50ca0.bin
```
The `hppa11sw-c50ca0.bin` file is the original firmware I saved.

kuba2k2, 2023-11-02:
Hi,
I've just written a brand new flashing tool for RTL8710 - without relying on `rtltool.py` anymore. It seems to be stable now, and supports read-retry and hash checking too. The GUI/CLI is obviously the same (ltchiptool), but it's different under-the-hood.

You can get it here - it's not released yet:
https://github.com/libretiny-eu/ltchiptool/tree/feature/flasher-update

protectivedad, 2023-11-02:
I opened up one of the devices to test but I get the same error:
```
S: <- TX: B1 26 48 E3 F7 4A F9 A0  78 00 90 E0 78 01 90 20  .&H..J..x...x.. 
S: <- TX: 79 02 90 60 79 03 90 21  48 A1 79 22 78 63 78 D0  y..`y..!H.y"xcx.
S: <- TX: F8 8C 01 E3 F7 3A F9 5A                           .....:.Z
S: -> RX: b'\x15'
E: ValueError: Failed to write to 0x8000000
E: |-- File "/home/anthony/ltchiptool/ltchiptool/soc/ambz/flash.py", line 148, in flash_write_raw
```
I thought maybe it was a power supply issue so I hooked up a supply that would power the device but got the same error.

closed 2023-11-02 by protectivedad

kuba2k2, 2023-11-02:
As I said in my previous comment, the updated version is not released yet. You need to clone the branch that I linked. The error you posted originates from the old tool, so your program wasn't updated.

protectivedad, 2023-11-02:
I did:
```
commit 057f4512d1ba16f4700db20819cf877eb011144f (HEAD -> master, tag: v4.7.0, origin/master, origin/HEAD)
Author: Kuba Szczodrzyński <kuba@szczodrzynski.pl>
Date:   Mon Oct 30 11:38:05 2023 +0100

    [release] v4.7.0
```
Is that the right one?

Cossid, 2023-11-02:
No, that is the master branch, you need to check out the `feature/flash-update` branch.

protectivedad, 2023-11-02:
Ah! That explains it. I looked at the link but I updated later on and must have switched to the master. Sorry for the trouble.

MiguelAReis, 2023-11-14:
Is anyone able to build this branch into an exe file? Thanks