libretiny-eu/ltchiptool #14: Error running ltchiptool gui under macOS 13.6.1

cohm, 2023-11-12:
I'm trying to use `ltchiptool` as part of following [this guide](https://digiblur.com/2023/08/19/updated-tuya-cloudcutter-with-esphome-bk7231-how-to-guide/) but I'm having trouble getting `ltchiptool` to run on my Mac (I see it's "untested" according to [the docs](https://docs.libretiny.eu/docs/flashing/tools/ltchiptool/), so perhaps this is useful feedback). FWIW, I'm on macOS 13.6.1 (22G313), I have python 3.11.6 via homebrew, and I installed `wxPython` and `ltchiptool`, but got this error when starting the gui. I tried both 4.2.3 as used in the guide, and then the current latest version (4.7.0), and got two different errors. The full output is below.

Any ideas how to get the GUI to work under macOS?

Thanks a lot in advance!

```
OhmBookPro13:~/Downloads > pip3 install wxpython
Collecting wxpython
  Using cached wxPython-4.2.1-cp311-cp311-macosx_10_10_universal2.whl.metadata (2.9 kB)
Requirement already satisfied: pillow in /usr/local/lib/python3.11/site-packages (from wxpython) (10.1.0)
Requirement already satisfied: six in /usr/local/lib/python3.11/site-packages (from wxpython) (1.16.0)
Requirement already satisfied: numpy in /usr/local/lib/python3.11/site-packages (from wxpython) (1.26.1)
Using cached wxPython-4.2.1-cp311-cp311-macosx_10_10_universal2.whl (31.5 MB)
Installing collected packages: wxpython
Successfully installed wxpython-4.2.1
OhmBookPro13:~/Downloads > pip3 install ltchiptool==4.2.3
Collecting ltchiptool==4.2.3
  Using cached ltchiptool-4.2.3-py3-none-any.whl.metadata (3.8 kB)
Requirement already satisfied: bitstruct<9.0.0,>=8.17.0 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (8.19.0)
Requirement already satisfied: bk7231tools<2.0.0,>=1.3.6 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (1.4.0)
Requirement already satisfied: click<9.0.0,>=8.1.3 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (8.1.7)
Requirement already satisfied: colorama<0.5.0,>=0.4.5 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (0.4.6)
Requirement already satisfied: hexdump<4.0,>=3.3 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (3.3)
Requirement already satisfied: importlib-metadata<5.0.0,>=4.12.0 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (4.13.0)
Requirement already satisfied: prettytable<4.0.0,>=3.3.0 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (3.9.0)
Requirement already satisfied: py-datastruct<0.5.0,>=0.4.0 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (0.4.0)
Requirement already satisfied: pycryptodomex<4.0.0,>=3.15.0 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (3.19.0)
Requirement already satisfied: semantic-version<3.0.0,>=2.10.0 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (2.10.0)
Requirement already satisfied: xmodem<0.5.0,>=0.4.6 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from ltchiptool==4.2.3) (0.4.7)
Requirement already satisfied: pyserial<4.0,>=3.5 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from bk7231tools<2.0.0,>=1.3.6->ltchiptool==4.2.3) (3.5)
Requirement already satisfied: zipp>=0.5 in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from importlib-metadata<5.0.0,>=4.12.0->ltchiptool==4.2.3) (3.17.0)
Requirement already satisfied: wcwidth in /Users/cohm/Library/Python/3.11/lib/python/site-packages (from prettytable<4.0.0,>=3.3.0->ltchiptool==4.2.3) (0.2.9)
Using cached ltchiptool-4.2.3-py3-none-any.whl (159 kB)
Installing collected packages: ltchiptool
Successfully installed ltchiptool-4.2.3
OhmBookPro13:~/Downloads > python3 -m ltchiptool gui
22:34:42: Debug: Adding duplicate image handler for 'Windows bitmap file'
22:34:42: Debug: Adding duplicate animation handler for '1' type
22:34:42: Debug: Adding duplicate animation handler for '2' type
Fatal Python error: PyGILState_Release: thread state 0x105888d28 must be current when releasing
Python runtime state: initialized

Current thread 0x00007ff84f543700 (most recent call first):
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/gui/colors.py", line 128 in apply
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/gui/panels/log.py", line 206 in OnPaletteChanged
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/gui/main.py", line 324 in palette
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/gui/main.py", line 187 in SetSettings
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/gui/main.py", line 242 in OnShow
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/util/../../ltchiptool/gui/__main__.py", line 30 in gui_entrypoint
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/util/../../ltchiptool/gui/__main__.py", line 48 in cli
  File "/Users/cohm/Library/Python/3.11/lib/python/site-packages/click/core.py", line 783 in invoke
  File "/Users/cohm/Library/Python/3.11/lib/python/site-packages/click/core.py", line 1434 in invoke
  File "/Users/cohm/Library/Python/3.11/lib/python/site-packages/click/core.py", line 1688 in invoke
  File "/Users/cohm/Library/Python/3.11/lib/python/site-packages/click/core.py", line 1078 in main
  File "/Users/cohm/Library/Python/3.11/lib/python/site-packages/click/core.py", line 1157 in __call__
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/__main__.py", line 112 in cli
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/__main__.py", line 119 in <module>
  File "<frozen runpy>", line 88 in _run_code
  File "<frozen runpy>", line 198 in _run_module_as_main

Extension modules: wx._core, wx._adv, wx._xml, wx._xrc, wx._dataview (total: 5)
Abort trap: 6
OhmBookPro13:~/Downloads > python3 --version
Python 3.11.6
OhmBookPro13:~/Downloads > pip3 uninstall ltchiptool       
Found existing installation: ltchiptool 4.2.3
Uninstalling ltchiptool-4.2.3:
  Would remove:
    /usr/local/bin/ltchiptool
[... 42 lines cut ...]
  File "/usr/local/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/threading.py", line 1002 in _bootstrap

Current thread 0x00007ff84f543700 (most recent call first):
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/gui/colors.py", line 128 in apply
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/gui/panels/log.py", line 247 in OnPaletteChanged
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/gui/main.py", line 334 in palette
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/gui/main.py", line 194 in SetSettings
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/gui/main.py", line 252 in OnShow
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/util/../../ltchiptool/gui/__main__.py", line 32 in gui_entrypoint
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/util/../../ltchiptool/gui/__main__.py", line 50 in cli
  File "/Users/cohm/Library/Python/3.11/lib/python/site-packages/click/core.py", line 783 in invoke
  File "/Users/cohm/Library/Python/3.11/lib/python/site-packages/click/core.py", line 1434 in invoke
  File "/Users/cohm/Library/Python/3.11/lib/python/site-packages/click/core.py", line 1688 in invoke
  File "/Users/cohm/Library/Python/3.11/lib/python/site-packages/click/core.py", line 1078 in main
  File "/Users/cohm/Library/Python/3.11/lib/python/site-packages/click/core.py", line 1157 in __call__
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/__main__.py", line 122 in cli
  File "/usr/local/lib/python3.11/site-packages/ltchiptool/__main__.py", line 129 in <module>
  File "<frozen runpy>", line 88 in _run_code
  File "<frozen runpy>", line 198 in _run_module_as_main

Extension modules: wx._core, wx._adv, wx._xml, wx._xrc, zeroconf._utils.time, zeroconf._dns, zeroconf._protocol.outgoing, zeroconf._protocol.incoming, zeroconf._cache, zeroconf._record_update, zeroconf._updates, zeroconf._handlers.record_manager, zeroconf._services, zeroconf._services.info, zeroconf._services.registry, zeroconf._listener, zeroconf._handlers.answers, zeroconf._handlers.multicast_outgoing_queue, zeroconf._history, zeroconf._handlers.query_handler, zeroconf._services.browser, wx._dataview (total: 22)
Abort trap: 6
OhmBookPro13:~/Downloads > /usr/local/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/multiprocessing/resource_tracker.py:254: UserWarning: resource_tracker: There appear to be 3 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '
```

kuba2k2, 2023-11-13:
It seems that MacOS doesn't support setting text styles. Try commenting out this line to see if a different error shows up:
https://github.com/libretiny-eu/ltchiptool/blob/master/ltchiptool/gui/colors.py#L128

kuba2k2, 2023-11-13:
I don't know what `Illegal instruction: 4` might mean. However, the web interface is NOT esphome-kickstart, which upk2esphome requires. Flash kickstart to the device and try upk2esphome again - maybe it will work if it's able to connect (it should even detect the Kickstart device's IP automatically).

I'm glad it works, I haven't seen it run on MacOS before but honestly it looks fantastic.

I wonder why the text style crashed the app - you can clearly see text style being applied in the log window (green text).

It's also good to know that downloading plugins actually works. On Linux, for example, switching to the Plugins tab crashes the program right away... One can obviously install plugins with `pip`. Oh and the "Download plugins..." button is actually there to *search* for plugins - it will display a list of all available :smile:

cohm, 2023-11-13:
Wow - thanks for the lightning reply! 

Ok, so my failed attempt is likely the reason, good to know. Should I just grab the .bin from here and upload via the web interface? https://github.com/tuya-cloudcutter/tuya-cloudcutter/blob/main/custom-firmware/ESPHome-Kickstart-v23.08.29_bk7231n_app.ota.ug.bin (I know my device has bk7231n)

I'm quite new to this, and only tried to upload a .uf2 file made by ESPhome - I don't want to brick the device by uploading a bad file..

About the text style, no idea - I *did* comment out that line (this was what allowed the gui to work on my mac), so I guess it's somehow not needed?

Interesting that the "Download plugins..." button should list available plugins - that's what I expected too, but I played around a bit since that actually didn't work for me. I got an error message saying "`request` doesn't exist" or similar. Now when I tried again to reproduce it and provide a screen dump, it actually worked :)

[image]

kuba2k2, 2023-11-13:
That's the cool part here - you can't brick the device by uploading the wrong file, unless you intend to do so by creating a bad file (or the firmware has problems booting up). For LibreTiny, you only need UF2 files - they work nearly everywhere.

That line you commented was used for *changing* style of the log window (if you change the color scheme in menu). The line responsible for setting colors in the first place is still somewhere there.

Oh so I forgot to include "requests" in dependencies.. good to know, let's keep this issue open. It works now because installing upk2esphome probably added the missing package.

cohm, 2023-11-13:
Great to hear about the brick protection! Sorry if I'm slow, but where do I get the kickstart uf2 file I should upload to see if I can get the grabbing via upk2esphome to work?

What you write about the text style and missing requests dependency makes perfect sense - happy to try to navigate through this on a Mac and hopefully provide some helpful feedback in the process :)

kuba2k2, 2023-11-13:
Get the kickstart file from esphome-kickstart repository. It's under the libretiny-eu organization on GitHub.

cohm, 2023-11-13:
Ok, progress - I grabbed the uf2 file (for anyone as inexperienced as me landing here, it can be found in the [assets of the release](https://github.com/libretiny-eu/esphome-kickstart/releases), but not in the repo itself where I first poked around). This uploaded just fine to the device via the web interface, and allowed the grabbing to start at least, even if it failed and killed the gui before it finished:
[image]
No obvious error in the web interface log there, but in the terminal:
```
OhmBookPro13:~/Downloads > python3 -m ltchiptool gui
23:22:03: Debug: Adding duplicate image handler for 'Windows bitmap file'
23:22:03: Debug: Adding duplicate animation handler for '1' type
23:22:03: Debug: Adding duplicate animation handler for '2' type
I: Loaded settings from /Users/cohm/Library/Application Support/ltchiptool/gui.json
I: Found new device: cu.BLTH - n/a
I: Found new device: cu.Bluetooth-Incoming-Port - n/a
Abort trap: 6
```
In case it's helpful, these were the last lines in the web interface log:
```
23:22:20	[D]	[hub_api:037]	
Reading flash: offset=1fd000, length=2048
23:22:20	[D]	[hub_api:037]	
Reading flash: offset=1fd800, length=2048
23:22:20	[D]	[hub_api:037]	
Reading flash: offset=1fe000, length=2048
23:22:20	[D]	[hub_api:037]	
Reading flash: offset=1fe800, length=2048
23:22:20	[D]	[hub_api:037]	
Reading flash: offset=1ff000, length=2048
23:22:20	[D]	[hub_api:037]	
Reading flash: offset=1ff800, length=2048
```

kuba2k2, 2023-11-13:
It seems to have finished reading at least. It's probably related to the way bk7231tools currently extracts that data - it uses some weird go-to magic. I'm currently in the middle of rewriting that.

cohm, 2023-11-13:
Ok, I guess I'm stuck here for now then, but I'm not in a rush in any way. Thanks again for the incredible support and fast replies so far!

cohm, 2024-01-03:
FWIW, I just tried this again after updating to ltchiptool 4.10.1, but I still get the same error as in my last message above. I guess I should look for a new release of bk7231tools before I should expect the suspected issue above to be resolved? I currently have 1.5.4.

Happy help test something if that's useful!

jon-daemon, 2024-03-09:
I also get this on Mac OS 11.7
```
/usr/local/Cellar/python@3.11/3.11.5/Frameworks/Python.framework/Versions/3.11/lib/python3.11/multiprocessing/resource_tracker.py:224: UserWarning: resource_tracker: There appear to be 4 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '
```
It happened when I connected CEN to GND so probably when the chip was seen. Also it happened when I clicked `Grab from ESPHome-Kickstart` on another chip with ESPHome-Kickstart firmware flashed.

cohm, 2024-12-29:
Holiday couch time again, so gave this another shot. I'm now on an M3 MBP with macos 14.7.1 with most things installed via homebrew (Python 3.13.0), and tried installing latest versions of ltchiptool, wxpython (and zeroconf since it seemed to be needed) in a venv. I can get the gui up with this setup, and install the UPK2ESPHome plugin - here's the output in the terminal:
```
(venv) OhmBookPro14 ~/Documents/Private/Boende/Telefoncentralen/Smartahemgrejer/ltchiptool % python3 -m ltchiptool gui      
11:52:39 AM: Debug: Adding duplicate image handler for 'Windows bitmap file'
11:52:39 AM: Debug: Adding duplicate animation handler for '1' type
11:52:39 AM: Debug: Adding duplicate animation handler for '2' type
I: Found new device: cu.Bluetooth-Incoming-Port - n/a
I: Found new device: cu.wlan-debug - n/a
I: Searching 'ltchiptool' on PyPI...
I: Collecting UPK2ESPHome
I:   Downloading upk2esphome-1.13.0-py3-none-any.whl.metadata (1.2 kB)
I: Collecting PyYAML<7.0,>=6.0 (from UPK2ESPHome)
I:   Downloading PyYAML-6.0.2-cp313-cp313-macosx_11_0_arm64.whl.metadata (2.1 kB)
I: Collecting inquirer<4.0.0,>=3.1.3 (from UPK2ESPHome)
I:   Downloading inquirer-3.4.0-py3-none-any.whl.metadata (6.8 kB)
I: Requirement already satisfied: requests<3.0.0,>=2.31.0 in ./venv/lib/python3.13/site-packages (from UPK2ESPHome) (2.32.3)
I: Collecting blessed>=1.19.0 (from inquirer<4.0.0,>=3.1.3->UPK2ESPHome)
I:   Downloading blessed-1.20.0-py2.py3-none-any.whl.metadata (13 kB)
I: Collecting editor>=1.6.0 (from inquirer<4.0.0,>=3.1.3->UPK2ESPHome)
I:   Downloading editor-1.6.6-py3-none-any.whl.metadata (2.3 kB)
I: Collecting readchar>=4.2.0 (from inquirer<4.0.0,>=3.1.3->UPK2ESPHome)
I:   Downloading readchar-4.2.1-py3-none-any.whl.metadata (7.5 kB)
I: Requirement already satisfied: charset-normalizer<4,>=2 in ./venv/lib/python3.13/site-packages (from requests<3.0.0,>=2.31.0->UPK2ESPHome) (3.4.1)
I: Requirement already satisfied: idna<4,>=2.5 in ./venv/lib/python3.13/site-packages (from requests<3.0.0,>=2.31.0->UPK2ESPHome) (3.10)
I: Requirement already satisfied: urllib3<3,>=1.21.1 in ./venv/lib/python3.13/site-packages (from requests<3.0.0,>=2.31.0->UPK2ESPHome) (2.3.0)
I: Requirement already satisfied: certifi>=2017.4.17 in ./venv/lib/python3.13/site-packages (from requests<3.0.0,>=2.31.0->UPK2ESPHome) (2024.12.14)
I: Requirement already satisfied: wcwidth>=0.1.4 in ./venv/lib/python3.13/site-packages (from blessed>=1.19.0->inquirer<4.0.0,>=3.1.3->UPK2ESPHome) (0.2.13)
I: Requirement already satisfied: six>=1.9.0 in ./venv/lib/python3.13/site-packages (from blessed>=1.19.0->inquirer<4.0.0,>=3.1.3->UPK2ESPHome) (1.17.0)
I: Collecting runs (from editor>=1.6.0->inquirer<4.0.0,>=3.1.3->UPK2ESPHome)
I:   Downloading runs-1.2.2-py3-none-any.whl.metadata (10 kB)
I: Collecting xmod (from editor>=1.6.0->inquirer<4.0.0,>=3.1.3->UPK2ESPHome)
I:   Using cached xmod-1.8.1-py3-none-any.whl.metadata (1.8 kB)
I: Downloading upk2esphome-1.13.0-py3-none-any.whl (76 kB)
I: Downloading inquirer-3.4.0-py3-none-any.whl (18 kB)
I: Downloading PyYAML-6.0.2-cp313-cp313-macosx_11_0_arm64.whl (171 kB)
I: Using cached blessed-1.20.0-py2.py3-none-any.whl (58 kB)
I: Downloading editor-1.6.6-py3-none-any.whl (4.0 kB)
I: Downloading readchar-4.2.1-py3-none-any.whl (9.3 kB)
I: Downloading runs-1.2.2-py3-none-any.whl (7.0 kB)
I: Using cached xmod-1.8.1-py3-none-any.whl (4.6 kB)
I: Installing collected packages: xmod, readchar, PyYAML, blessed, runs, editor, inquirer, UPK2ESPHome
I: Successfully installed PyYAML-6.0.2 UPK2ESPHome-1.13.0 blessed-1.20.0 editor-1.6.6 inquirer-3.4.0 readchar-4.2.1 runs-1.2.2 xmod-1.8.1
I: Installed plugin upk2esphome
```
So far so good:
[image]
But when trying to grab from the found device, I get a hard crash - this seems to be the relevant thread info:
[image]
This is all that's in the terminal:
```
(venv) OhmBookPro14 ~/Documents/Private/Boende/Telefoncentralen/Smartahemgrejer/ltchiptool % /opt/homebrew/Cellar/python@3.13/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/multiprocessing/resource_tracker.py:276: UserWarning: resource_tracker: There appear to be 4 leaked semaphore objects to clean up at shutdown: {'/mp-w5ae5_ed', '/mp-85wwwbdg', '/mp-k8v1ca7n', '/mp-q25oseud'}
  warnings.warn(
```

Just figured I'd report this in case it's helpful for @kuba2k2 or others watching!

ohmantics, 2025-03-06:
Following along and getting a different crash on Sequoia 15.3.1 when clicking "Grab from ESPHome-Kickstart":

> *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: '-[HIRunLoopSemaphore wait:] has been invoked on a secondary thread; the problem is likely in a much shallower frame in the backtrace'
> 
> *** First throw call stack:
> (
> 	0   CoreFoundation                      0x000000019d206e80 __exceptionPreprocess + 176
> 	1   libobjc.A.dylib                     0x000000019cceecd8 objc_exception_throw + 88
> 	2   Foundation                          0x000000019e416198 -[NSCalendarDate initWithCoder:] + 0
> 	3   HIServices                          0x00000001a3b2f154 -[HIRunLoopSemaphore wait:] + 340
> 	4   HIToolbox                           0x00000001a882f3b4 __62-[IMKInputSessionXPCInvocation_Modern invocationAwaitXPCReply]_block_invoke + 72
> 	5   HIToolbox                           0x00000001a882ff10 +[IMKInputSession_Modern withActivity:performActions:] + 440
> 	6   HIToolbox                           0x00000001a882f35c -[IMKInputSessionXPCInvocation_Modern invocationAwaitXPCReply] + 116
> 	7   HIToolbox                           0x00000001a88314d0 -[IMKInputSession_Modern deactivate] + 876
> 	8   HIToolbox                           0x00000001a87bfb7c IMKInputSessionDeactivate + 40
> 	9   HIToolbox                           0x00000001a87bee40 DeactivateInputMethodInstance + 68
> 	10  HIToolbox                           0x00000001a87a7444 utDeactivateAllSelectedIMInDocIterator + 96
> 	11  CoreFoundation                      0x000000019d155b74 CFArrayApplyFunction + 72
> 	12  HIToolbox                           0x00000001a87a73cc utDeactivateAllSelectedIMInDoc + 148
> 	13  HIToolbox                           0x00000001a87a215c MyDeactivateTSMDocument + 380
> 	14  HIToolbox                           0x00000001a87a2540 MyActivateTSMDocument + 584
> 	15  AppKit                              0x00000001a0e9ae10 -[NSTextInputContext activate] + 480
> 	16  AppKit                              0x00000001a0d103f4 +[NSTextInputContext currentInputContext_withFirstResponderSync:] + 300
> 	17  AppKit                              0x00000001a0e44820 -[NSTextView(NSSharing) setEditable:] + 200
> 	18  libwx_osx_cocoau_core-3.2.0.3.0.dyl 0x00000001053a015c _ZN19wxNSTextViewControl9WriteTextERK8wxString + 120
> 	19  libwx_osx_cocoau_core-3.2.0.3.0.dyl 0x000000010532362c _ZN11wxTextEntry9WriteTextERK8wxString + 120
> 	20  _core.cpython-313-darwin.so         0x0000000105f99780 _ZL27meth_wxTextEntry_AppendTextP7_objectS0_S0_ + 152
> 	21  Python                              0x0000000102fcc5e4 cfunction_call + 76
> 	22  Python                              0x0000000102f6e424 _PyObject_MakeTpCall + 124
> 	23  Python                              0x0000000103096a94 _PyEval_EvalFrameDefault + 9684
> 	24  Python                              0x0000000102f71c7c method_vectorcall + 396
> 	25  Python                              0x0000000102f6e0c8 PyObject_CallNoArgs + 88
> 	26  Python                              0x0000000103094994 _PyEval_EvalFrameDefault + 1236
> 	27  Python                              0x0000000102f71c7c method_vectorcall + 396
> 	28  Python                              0x000000010318d2c4 thread_run + 152
> 	29  Python                              0x000000010311928c pythread_wrapper + 48
> 	30  libsystem_pthread.dylib             0x000000019d0ac2e4 _pthread_start + 136
> 	31  libsystem_pthread.dylib             0x000000019d0a70fc thread_start + 8
> )
> libc++abi: terminating due to uncaught exception of type NSException
> [1]    37708 abort      ltchiptool gui
> /opt/homebrew/Cellar/python@3.13/3.13.1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/multiprocessing/resource_tracker.py:276: UserWarning: resource_tracker: There appear to be 4 leaked semaphore objects to clean up at shutdown: {'/mp-9wtv686v', '/mp-297u_ko1', '/mp-42a9ue5p', '/mp-1ips5fbx'}
>   warnings.warn(

ohmantics, 2025-03-06:
Additional notes:

- Download Plugins button doesn't work (as noted above by @cohm )
- CLI is able to pull from kickstart just fine.