libretiny-eu/ltchiptool #16: Unable to convert some datapoints.

hackintoshlover, 2023-12-28:
Hi,
Thanks for the amazing software.
I have a 2 channel tuya device with built-in rf module. https://www.primezen.in/products/zen-2c5w/
I successfully extracted the firmware from the device. 
While analyzing using ltchiptool, it says tuyamcu found and needs to download schema.
It downloads a schema which is unable to convert few datapoints.
Attached is an image for better understanding.
[image]

3 datapoints viz,
102 - sends a pairing command to internal mcu which puts rf device in pairing mode to pair a rf switch.
103 - sends a clear pairing command to internal mcu which clears pairing information from rf device.
116 - switches on both relays at once.

I have two questions.
1) How do i make above three work.
2) How to get new schema since the schema used by ltchiptool to poll data differs from the schema available in my device as stated in the image.

Attached below is the firmware binary for your reference.
[ltchiptool_dump_2023-12-20_23-13-00.zip](https://github.com/libretiny-eu/ltchiptool/files/13788295/ltchiptool_dump_2023-12-20_23-13-00.zip)

Kindly help.
Thanks.

kuba2k2, 2023-12-28:
1. You need to implement these manually into your YAML. Not all datapoint types are supported, among those are write-only enum types (as in your example).
2. ltchiptool pulls whatever schema is returned for the device by Tuya servers. The mismatching ID likely only means that it pulled an updated version of the schema.

hackintoshlover, 2023-12-28:
1. Ok, I'll try that and get back to you.
2. Is there a manual way of pointing out schema. example from my tuya developer account?

kuba2k2, 2023-12-28:
No, there isn't. Usually the schemas aren't any different anyway, so it would most likely produce the same output.

Keep in mind that UPK2ESPHome is only meant to be a starting point - you should always review the generated output and adjust to your needs.

hackintoshlover, 2023-12-28:
Will keep that in mind.
Thanks.

closed 2023-12-28 by hackintoshlover