libretiny-eu/ltchiptool #18: Help with a Midea SmartKit Dongle

dhewg, 2024-01-25:
My Midea dehumidifier came with a WiFi dongle, the case reads `EU-SK-105`.
It has a USB connector, but the data lines are just UART TTL lines to another board. Those are apparently 5V.

It's rather tiny, not that readable and I don't have a nice cam around, but apparently it has a RTL8720C:
[image]
[image]
[image]

If I attach a 3.3V CH341 UART TTL to LOG_TX/RX I can read from and write to it:
```
== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0

== Boot Loader ==
Nov 30 2020:20:11:23

Boot Loader <==

== RAM Start ==
Build @ 17:50:15, Dec  2 2022

$8710c>
Initializing WIFI ...
WIFI initialized
count:15, interval:8

init_thread(57), Available heap 0x154c0

#
#
#help

WLAN AT COMMAND SET:
==============================
1. Wlan Scan for Network Access Point
   # ATWS
2. Connect to an AES AP
   # ATW0=SSID
   # ATW1=PASSPHRASE
   # ATWC
3. Create an AES AP
   # ATW3=SSID
   # ATW4=PASSPHRASE
   # ATW5=CHANNEL
   # ATWA
4. Ping
   # ATWI=xxx.xxx.xxx.xxx

```

(note that the initial prompt reads `8710c`)

But I'm not sure how to enter download mode, the `flash info ambz2` help reads `strapping pin (GPIO 0 / PA00) has to be pulled *to 3.3V*`. Since that a current and I try to salvage that thing instead of destroying it.... any idea which contacts I have to short?

Thanks!

kuba2k2, 2024-01-25:
First of all, I don't see a 3.3V regulator on this board, nor any transistors required to convert UART to the apparent 5V. It's most likely just misleading silkscreen - **DO NOT** power it from 5V, you may kill the chip!

I don't know which pin might be A0. You might have a bit of luck by checking the RTL8720CF pinout and seeing where the PCB trace goes from that pin.

Since the device uses default AT firmware, it's totally possible that the A0 pin is just not there at all. Or maybe the firmware has a command to run download mode - the AT command set printed with `help` is just a small part of the available commands, IIRC.

dhewg, 2024-01-25:
Ok, thanks, I'll try to trace...

I found a few AT commands, does any of those ring any bell?
```
unknown command 'AT'
unknown command 'AT?'
[ATSR]: _AT_SYSTEM_RECOVER_OTA_SIGNATURE_
unknown command 'ATSV'
ATSP
[ATSE]: _AT_SYSTEM_EDIT_REGISTER_
[ATSE] Usage: ATSE=REGISTER[VALUE]
unknown command 'ATSY'
unknown command 'ATSU'
unknown command 'ATSO'
[ATSC]: _AT_SYSTEM_CLEAR_OTA_SIGNATURE_
[ATSG]: _AT_SYSTEM_GPIO_TEST_
[ATSG] Usage: ATSG=PINNAME(ex:A0)
```

dhewg, 2024-01-26:
Found `ATXX` in `packages/framework-realtek-ambz2/component/common/api/at_cmd/atcmd_sys.c`

```
#ATXX

== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0
Test Mode: boot_cfg1=0x20
Download Image over UART2[tx=16,rx=15] baud=115200
```

```
$ ltchiptool flash info -d /dev/ttyUSB0 ambz2
I: Connecting to 'Realtek AmebaZ2' on /dev/ttyUSB0 @ 115200
I: |-- Success! Chip info: RTL8720CF
I: Reading chip info...
I: Chip: RTL8720CF
I: +---------------------+-------------------+
I: | Name                | Value             |
I: +---------------------+-------------------+
I: | Chip VID            | 5                 |
I: | Chip Version        | 3                 |
I: | ROM Version         | v3.0              |
I: |                     |                   |
I: | Chip Type           | RTL87x0CF         |
I: | MAC Address (Wi-Fi) | FF:FF:FF:FF:FF:FF |
I: | MAC Address (BT)    | FF:FF:FF:FF:FF:FF |
I: | Boot Debugging      | Disabled          |
I: | Secure Boot         | Disabled          |
I: |                     |                   |
I: | Flash Type          | RTL8720CF         |
I: | Flash Mode          | SINGLE            |
I: +---------------------+-------------------+
I: |-- Finished in 0.834 s
```

Nice!

kuba2k2, 2024-01-26:
Great. Keep in mind that if you flash something that doesn't work, you might never be able to enter flashing mode again, because of no A0 pin.

dhewg, 2024-01-26:
Good point. It might still be there, but it's all so tiny and I don't have proper tools for that

dhewg, 2024-01-26:
And with that in mind, just to double check, does this look okay to you? Just to get an OTA ready esphome on there:
```
esphome:
  name: dehumidifier

rtl87xx:
  board: generic-rtl8720cf-2mb-992k

logger:
  level: VERBOSE

api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    password: !secret wifi_ap_password

captive_portal:
```

I got a few warnings with that:
```
.platformio/packages/library-freertos/FreeRTOS/Source/tasks.c:2193:6: warning: implicit declaration of function '__get_IPSR' [-Wimplicit-function-declaration]
...
Linking .pioenvs/dehumidifier/raw_firmware.elf
|-- Image 1: raw_firmware.ota1.elf
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .ram.code_text changed by 4
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .glue_7 changed by 2
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .glue_7t changed by 2
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .vfp11_veneer changed by 2
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .v4_bx changed by 2
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .iplt changed by 2
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .system_restore_data changed by 8
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .bss changed by 32
|-- Image 2: raw_firmware.ota2.elf
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .ram.code_text changed by 4
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .glue_7 changed by 2
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .glue_7t changed by 2
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .vfp11_veneer changed by 2
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .v4_bx changed by 2
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .iplt changed by 2
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .system_restore_data changed by 8
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: warning: start of section .bss changed by 32
```

kuba2k2, 2024-01-26:
ESPHome doesn't support OTA for RTL8720C. **You won't be able to update via OTA yet**, however you should be able to flash via UART in most cases - LibreTiny should automatically reboot to download mode when you start flashing.

Note that this is not stable software yet. Rebooting to download mode might not work and you'll have trouble getting the board flashed again.

dhewg, 2024-01-26:
Oh okay. Maybe it's a good idea to error out on `ota:` then?
Is there any mechanism to enter download mode? With esphome's boot loop detection? Spam some key on uart tx upon boot?

kuba2k2, 2024-01-26:
`ota:` can stay, it will do no wrong. Uploading any file won't work anyway.

Download mode should enter automatically when you connect the flasher program. It detects a specific byte sequence.

See: https://docs.libretiny.eu/docs/flashing/tools/adr/?h=auto

dhewg, 2024-01-26:
Of course it didn't work...

```
== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0

== Boot Loader ==
Nov 30 2020:20:11:23
[MISC Err]Sub-Image FST Decrypt Err!
Boot Load Err!
```

I'll fetch my equipment from the office and try to trace later. Maybe I can find A0, I do have a full flash backup

dhewg, 2024-12-05:
Nope.

It's been a while, but I barely spent time on it since my last message.

But if you figure something out please let us know!

panzerknacker20111, 2025-05-30:
Hello guys,

I found the A0 Pin a while ago and today found this issue.
So here is what I got: 

A0 is routed to the 2nd Pin on the left and there is an +3.3V Pin on the PCB. I just soldered an copper wire from the +3.3V to A0. 

[image]

It worked right away. Interestingly I had to connect the pcb to voltage and than connect the Usb->Serial adapter. 
Maybe when the Adapter was connected when powering on it did some inteference. IDK. But in the End I was able to pull an Full Backup of the Flash and ROM. So I think that the A0 Mode is working as expected.

If your intend is to use it later with esphome I can only give you the advice to just buy one of those esp8266 usb dongles made for those Midea type AC units. It saves you a lot of trouble and you can leave this oem dongle as it is. 

But which way you choose is of course your own decision

sammothxc, 2026-10-11:
@skopa here's that stock firmware from my US-SK105: 
[https://gist.github.com/sammothxc/a49dec4f0f1524fe16bfbd26cac045e5](https://gist.github.com/sammothxc/a49dec4f0f1524fe16bfbd26cac045e5)

@dhewg and @panzerknacker20111, thanks a ton for your help locating A0. Because the UART firmware loader is in ROM, its nearly impossible to brick these things, you just hold A0 high at boot and you can just keep flashing it.

It took me maybe 15 minutes to edit 2 or 3 files in the Midea component and 2 flashing attempts, but I got this thing working great with esphome! A few bucks saved is a few bucks saved.

timmerk, 2026-10-11:
Interestingly, the newer wall thermostat control boxes come with one of these (well, a US- SK107, but looks pretty much the same) soldered onto the main PCB:

[image]

I might have to try it this on that!

sammothxc, 2026-10-11:
@timmerk wow that board looks nearly identical, just connected via the castellated pins instead of USB. It also has that metal housing but all the necessary pins should be exposed. I'm putting together a repo of my edits:

https://github.com/sammothxc/midea-sk105-esphome

I'd be interested to see if this works for more of the SK10x family.