libretiny-eu/ltchiptool #21: Chip Type - Unrecognized (0x48405F31)

diepeterpan, 2024-02-26:
I bought some switches with CB2S chips and BK7231N's on them but am unable to flash them.

Trying to ID the chips it sends back the following.....

```
ltchiptool -v flash info bk72xx

: |-- Success! Chip info: None
I: Reading chip info...
I: Chip: None
I: +--------------------+---------------------------+
I: | Name               | Value                     |
I: +--------------------+---------------------------+
I: | Chip Type          | Unrecognized (0x48405F31) |
I: | Bootloader Version | Unspecified               |
I: | Protocol Type      | BASIC_BEKEN               |
I: +--------------------+---------------------------+
I: |-- Finished in 5.271 s
```
[image]

kuba2k2, 2024-02-26:
Are these switches based on Tuya cloud? Some other manufacturers use different encryption keys sometimes.

Can you dump the firmware first? It would help to find what kind of bootloader is there.

diepeterpan, 2024-02-26:
I used the Android Tuya app so I guess it must be Tuya.  The box is very generic and says "Tuya Basic Wi-Fi Switch/Breaker".

I am unable to dump the firmware....

```
**ltchiptool -v flash read bk7231n output** 
D: Found Local data snapshot (v1.5.0) - /usr/local/lib/python3.11/site-packages/ltchiptool
I: Available COM ports:
I: |-- ttyUSB0 - FT232R USB UART - FT232R USB UART - FTDI (0403/6001)
I: |   |-- Selecting this port. To override, use -d/--device
I: |-- ttyS0 - ttyS0
I: |-- ttyS1 - ttyS1
I: Connecting to 'Beken 7231N' on /dev/ttyUSB0 @ 115200
D: Linking: stage 0
D: Linking: stage 1
D: Linking: stage 2
D: Linking: stage 3
I: Connect UART1 of the BK7231 to the USB-TTL adapter:
I: 
I:     --------+        +--------------------
I:          PC |        | BK7231             
I:     --------+        +--------------------
I:          RX | ------ | TX1 (GPIO11 / P11) 
I:          TX | ------ | RX1 (GPIO10 / P10) 
I:             |        |                    
I:         GND | ------ | GND                
I:     --------+        +--------------------
I:  
I: Using a good, stable 3.3V power supply is crucial. Most flashing issues
I: are caused by either voltage drops during intensive flash operations,
I: or bad/loose wires.
I:  
I: The UART adapter's 3.3V power regulator is usually not enough. Instead,
I: a regulated bench power supply, or a linear 1117-type regulator is recommended.
I:  
I: To enter download mode, the chip has to be rebooted while the flashing program
I: is trying to establish communication.
I: In order to do that, you need to bridge CEN pin to GND with a wire.
I: |-- Success! Chip info: None
I: Reading Flash (2 MiB) to 'output'
  [----------------------------------------------------------------]    0%D: Reading 4k page at 0x200000 (0.00%)
Reading failure (Incomplete response read: 3534 != 4101), retrying (attempt 0)
Reading failure (No response received), retrying (attempt 1)
Reading failure (No response received), retrying (attempt 2)
Reading failure (No response received), retrying (attempt 3)
Reading failure (No response received), retrying (attempt 4)
Reading failure (No response received), retrying (attempt 5)
Reading failure (No response received), retrying (attempt 6)
Reading failure (No response received), retrying (attempt 7)
Reading failure (No response received), retrying (attempt 8)
Reading failure (No response received), retrying (attempt 9)
Reading failure (No response received), retrying (attempt 10)
Reading failure (No response received), retrying (attempt 11)
```

diepeterpan, 2024-02-26:
Thank you, I have tried an external power source but no change, also checked the soldering again. I will try and troubleshoot. Maybe somebody else experiences a similar problem and can help.

diepeterpan, 2024-07-12:
After a couple of months finally got back to this, I did more research and found the CB2S chips in my devices were incorrectly marked on the silkscreen PCB, the entire row is reversed.  It seems like I am not the only one - https://www.elektroda.com/rtvforum/topic4042634.html.  Anyway, after soldering 3V3 to P26, GND to P24, and flipping TX and RX, all flashed correctly.

Seems like chips marked as 2317 100 are the problem.

[image]

kuba2k2, 2024-07-12:
For future reference, please use docs.libretiny.eu for pinouts. They have been checked and verified by many people to be correct. They also make it hard to mistake the front/back side of the PCB.