libretiny-eu/ltchiptool #22: Firmware Dump Analysis

downriverbob, 2024-03-09:
I have a firmware file I have read from a device running RTL8710, but it can't be analyzed with ltchiptool.
I receive the following error:

`E: UPK: The chosen device doesn't contain pin configuration.

Possible causes:
- it has vendor-specific firmware
- it uses TuyaMCU (report error if that's the case!)

Auto-generating ESPHome YAML is not possible.
`

Which is fine, makes sense because this device is unable to be flashed using tuya-cloudcutter.

I have flashed it with a blank config and am able to access the web server, so OTA updates via ESPHome should be possible.

How can the firmware be analyzed to find the correct configuration for this device?

Cossid, 2024-03-09:
Dump Analysis currently only works with Beken chips.  You can post the dump to the #dump channel in the LibreTiny discord channel, and we can see if it is something that is available or not, but there is currently not enough information to attempt to adapt the tools to do this automatically, as the sample of Realtek dumps is still quite small.

downriverbob, 2024-03-09:
Perfect, thanks. File has been posted

closed 2024-03-09 by downriverbob