libretiny-eu/ltchiptool #24: Unrecognized chip 0x540A8B75

Omantik, 2024-03-17:
Hello
Sber lamp at CBLC9
[image]
[image]

kuba2k2, 2024-03-17:
Hi, please read:
https://docs.libretiny.eu/docs/platform/beken-72xx/keys/
This explains why you're getting the error message.

For dumping the firmware and flashing new one, try using this version of bk7231tools:
https://github.com/tuya-cloudcutter/bk7231tools/tree/feature/flash-refactor
It should correctly identify the chip. Please post a screenshot of the chip info window. If you can, a ROM dump and a firmware dump would be appreciated too.

Omantik, 2024-03-17:
[Uploading 2 lamps.zip…]()

Omantik, 2024-03-17:
dump
[readResult_BK7231N_QIO_2343_2024-28-1-10-44-05 sber e14.zip](https://github.com/libretiny-eu/ltchiptool/files/14627118/readResult_BK7231N_QIO_2343_2024-28-1-10-44-05.sber.e14.zip)

Omantik, 2024-03-17:
dump2 from other lamp
[readResult_BK7231N_QIO_2024-17-3-11-58-59 sber e14-2.zip](https://github.com/libretiny-eu/ltchiptool/files/14627120/readResult_BK7231N_QIO_2024-17-3-11-58-59.sber.e14-2.zip)

kuba2k2, 2024-03-17:
Have you flashed anything to the device before dumping that firmware? It has odd data at 0x132000, which look like some firmware which shouldn't be there.

Omantik, 2024-03-17:
both dump genue from new lamps

Omantik, 2024-03-17:
I have another new lamp - I can still read from it

Omantik, 2024-03-17:
[e14-2.zip](https://github.com/libretiny-eu/ltchiptool/files/14627860/e14-2.zip)
first file from new lamp which never turned on
second file after ota update to last firmware

kuba2k2, 2024-03-17:
ESPHome config for these two devices:
```yaml
esphome:
  name: my-device
  platformio_options:
    board_build.bkcrypt_coeffs: 7d1ff9b3e3ffefde5d93b57eefefeb8b
    board_build.bkota.key: 908f771bac2ff1aa619b897c14c5a88b
    board_build.bkota.iv: 1e647fab598d727c
    board_flash.download: "0x132000+0xAE000"
```

Omantik, 2024-03-17:
friend, maybe i ask stupid question
i create simple yams and for esphome start
`esphome:
  name: "sber-e14"
  platformio_options:
    board_build.bkcrypt_coeffs: 7d1ff9b3e3ffefde5d93b57eefefeb8b
    board_build.bkota.key: 908f771bac2ff1aa619b897c14c5a88b
    board_build.bkota.iv: 1e647fab598d727c
    board_flash.download: "0x132000+0xAE000"

bk72xx:
  board: generic-bk7231n-qfn32-tuya

logger:

web_server:

captive_portal:

mdns:

api:
  reboot_timeout: 15min
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

text_sensor:
  - platform: libretiny
    version:
      name: LibreTiny Version

sensor:
  - platform: wifi_signal
    name: WiFi_Signal
  - platform: uptime
    name: uptime_sensor
`
same as UF2 - program vin ltchiptool and crc error
[image]

kuba2k2, 2024-03-17:
As I mentioned, you need to use the other branch of bk7231tools:
https://github.com/tuya-cloudcutter/bk7231tools/tree/feature/flash-refactor

First install ltchiptool using Python, with `pip install ltchiptool[gui]`. Then download this branch, open a terminal inside the downloaded directory, and run `pip install .`. Then start ltchiptool using `python -m ltchiptool gui`.

Omantik, 2024-03-17:
as i understand it not work at windows
tomorrow try install ubuntu.

kuba2k2, 2024-03-17:
It does work on Windows. If you install Python 3.10 or newer it will work on Windows.

Omantik, 2024-03-17:
ok thanks i try

Omantik, 2024-03-17:
i install python - do what you say
[image]
when program UF2 file again CRC error
[image]

kuba2k2, 2024-03-17:
You didn't run the `pip install .` command with downloaded bk7231tools repo. Make sure to download the `feature/flash-refactor` branch, not `master`.

Omantik, 2024-03-17:
Now the firmware is written successfully without crc errors, but lamp not start - not connect to wi-fi

kuba2k2, 2024-03-17:
You haven't added Wi-Fi AP to the config. Please do that and try again.
Also, in order to troubleshoot this, you'll need to connect UART to TX2 and view the device log output.

Omantik, 2024-03-17:
wifi session present with my wifi and password
i try recreate yaml and try reprogram again, but now receive it
[image]

kuba2k2, 2024-03-17:
Please post a screenshot of "get chip info" window. This version of bk7231tools should give more helpful information.

Omantik, 2024-03-17:
[image]

Omantik, 2024-03-17:
When i return stok - lamp start
After what i try again program UF2 - same error what locked

NonPIayerCharacter, 2024-03-18:
Worked for me, used OTA image and flashed at 0x132000, now it boots.

Omantik, 2024-03-18:
> Worked for me, used OTA image and flashed at 0x132000, now it boots.

plz more info how you to it. you create ota update file at esphome and program it from 0x132000?
[image]
this?

NonPIayerCharacter, 2024-03-18:
Yes

Omantik, 2024-03-18:
> Yes

Thx. at evening i try

NonPIayerCharacter, 2024-03-18:
[OpenBK7231N_App_1.0.0.zip](https://github.com/libretiny-eu/ltchiptool/files/14634076/OpenBK7231N_App_1.0.0.zip)
OpenBeken, if needed. Flash at 0x132000, doesn't support default OTA packages (need to build manually). Tried QIO, but it doesn't boot.

kuba2k2, 2024-03-18:
>  @kuba2k2 is there a way to build and flash 1.0.1 bootloader? And will it support unencrypted OTA?
No. The encryption keys are stored in eFuse and are not changeable.

NonPIayerCharacter, 2024-03-18:
> > @kuba2k2 is there a way to build and flash 1.0.1 bootloader? And will it support unencrypted OTA?
> 
> No. The encryption keys are stored in eFuse and are not changeable.

I am not talking about keys, but about bootloader itself. AFAIK there are bootloaders, which will allow OTA without encryption, and then flash it while encrypting. Not sure about that it is how it works. What about flashing bootloader? I built OpenBK7231N sdk with correct keys, and tried flashing both bk7231n_bootloader_enc.bin and QIO app at 0x0. Neither booted.

kuba2k2, 2024-03-18:
We don't have bootloader sources. All we have are binaries from either Beken SDKs or extracted from devices. The bootloader needs to know the eFuse encryption keys in order to run the OTA update. It is possible to modify an existing bootloader (say, 1.0.1) to change its encryption keys and enable unencrypted OTA, but it would require decompiling and modifying it, as well as packaging the bootloader binary to work with the device's keys.

An easier option is to simply pass the existing keys to LibreTiny in the ESPHome YAML, which generates a firmware binary that can just be flashed to 0x11000.

If that doesn't work, then either you're doing something wrong, or the encryption routines used in LibreTiny are somehow incorrect. It is believed that they could be wrong, because with a particular set of keys (a different SBER device) they were crashing, while working just fine on the device.

If you want us to try and fix it, please:
- compile ESPHome firmware with correct keys for your particular device
- grab the OTA and UF2 files from that particular build
- flash it using the manual OTA method (at 0x132000)
- dump the entire flash firmware (2 MiB)
- attach both the flash dump (2 MiB .bin) and the UF2 file

NonPIayerCharacter, 2024-03-18:
Understood. I thought, reading build.sh and seeing bootloader encryption, that it is built from source. LibreTiny OTA works normally, i thought that replacing stock bootloader will allow OpenBeken OTA to function, without manually rebuilding it from source.

kuba2k2, 2024-03-18:
I'm not familiar with the OpenBeken codebase. I don't know what solutions they use to deal with OTA and encryption.

I also don't know what device you're using. If you have another SBER device, OTA packages are most likely encrypted with a non-default key. LibreTiny allows to change the encryption keys, OpenBeken doesn't, AFAIK.

NonPIayerCharacter, 2024-03-18:
My device is the same as in the issue, SBER SBDV-00117 (LED C37 E14). Encryption keys are from your earlier reply

> ESPHome config for these two devices:
>
> ```yaml
> esphome:
>   name: my-device
>   platformio_options:
>     board_build.bkcrypt_coeffs: 7d1ff9b3e3ffefde5d93b57eefefeb8b
>     board_build.bkota.key: [...]

LibreTiny works OK with them, as does OpenBeken. But OpenBeken requires to be built manually, while replacing original tuya keys with sber ones, and OTA will work only with manually built binaries, not those, provided in github-actions.

kuba2k2, 2024-03-18:
Yes, as I mentioned, OpenBeken uses default keys and not SBER keys, and doesn't allow changing it. If you want to use OpenBeken, you're going to need to manually build it, or reach out to the project author to allow that.

Also note that the OTA key is not related to the encryption key (bkcrypt_coeffs). Your device might use the same OTA key but a different encryption key, in which case flashing UF2 will not work, but flashing manual OTA will.

NonPIayerCharacter, 2024-03-18:
bkcrypt_coeffs is probably the same. I erased my device fully, then flashed only bootloader and RF, and then flashed UF2. It worked normally.

Omantik, 2024-03-18:
> bkcrypt_coeffs is probably the same. I erased my device fully, then flashed only bootloader and RF, and then flashed UF2. It worked normally.

it work!!!! THANKS

closed 2024-03-19 by Omantik

tehb13, 2024-10-26:
@kuba2k2 please help with keys for flashing esphome
Sber smart relay [SBDV-00050](https://www.elektroda.com/rtvforum/viewtopic.php?p=21278069)
[DUMP_sbdv_00050.bin](https://github.com/user-attachments/files/17531980/readResult_BK7231N_QIO_sbdv_00050.bin.zip)

kuba2k2, 2024-10-27:
```yaml
esphome:
  name: sbdv-00050
  platformio_options:
    board_build.bkcrypt_coeffs: d34fbdd3efcbeadefd97bbffc7bfff3b
    board_build.bkota.key: b4f9937b63a7c365f27bf8d06fb67978
    board_build.bkota.iv: 53de9074383e2223
    board_flash.download: "0x132000+0xCC000"
```