libretiny-eu/ltchiptool #25: v4.11.0 throws errors ...

Another-Taipan, 2024-03-20:
When I try to use version 4.11.0 on Windows 11 to "Get Chip Info", this is the error I get:

[image]

The previous version 4.10.2 will "Get Chip Info" and "Read Flash", but it won't "Write Flash" - instead I get this error:

[image]

I am not having much luck, am I ...

Please let me know if you need more information?
Dennis

Cossid, 2024-03-20:
4.11.0 has a known issue that is still unresolved.
4.10.2 should work, especially for BK7231T(/S), so if you're getting no response received, it's probably worth trying again.

Another-Taipan, 2024-03-21:
I tried to get version 4.10.2 to "Write Flash" about 20 times without success.
So I tried version 4.4.1 (as used [here](https://www.youtube.com/watch?v=t0o8nMbqOSA&t=885s)) and it worked first time.  

Which makes it look even more likely that v4.10.2 has a bug preventing the Write Flash from ever working?

kuba2k2, 2024-03-21:
No. The "No response received" error is caused by timing issues with resetting the chip. Most likely you are resetting it too quickly or don't supply enough current from the power supply.

The reason that v4.4.1 worked might be because a lot of flash writing code has been improved in the meantime. Old versions might have different timing requirements (which just seems to match your resetting scheme), but they also have more bugs and other issues.

Therefore it's recommended to use the latest version - 4.10.2 (because 4.11.0 has a bug where it can't fully open) - and learn how to reset the chip properly. With BK7231T especially it takes a moment to get a grasp on how to do it.

Another-Taipan, 2024-03-21:
I appreciate that you know a lot more about this proceedure than I do, so I don't wish to argue with you as that would be very disrespectful.  

Except to say that the power supply I am using can provide up to 3A at 3.3V, so power should not be an issue.  And that I was able to "Get Chip Info" and "Read Flash" to work easily, which both involve resetting the chip.

I did notice that wording like "writing" flashed up briefly just before the "No response received" window popped up, so it appeared that I did reset the chip correctly and the writing did commence, but then failed?

Anyway, I have successfully installed the new ESPHome firmware and it appears to be working fine.  Thankyou for all your hard work with this tool.

kuba2k2, 2024-03-21:
v4.11.1 fixes the GUI launching issue. If you want, you can check if you're still experiencing problems with the latest release.