libretiny-eu/ltchiptool #26: RuntimeError: Error: 28841002: No permissions. Your subscription to cloud development plan has expired.

KyleStilkey, 2024-03-23:
I am unsure if the tool provides it's own keys but if so it looks like it has expired, if this is something I am supposed to update/change on my end I apologize but would like some steps to get the new License keys. Thanks!

kuba2k2, 2024-03-25:
Hi, can you check if it works now? I have tested it a few times and had mixed results. Sometimes different error codes showed up, maybe it was too early on after account creation.

KyleStilkey, 2024-03-29:
> Hi, can you check if it works now? I have tested it a few times and had mixed results. Sometimes different error codes showed up, maybe it was too early on after account creation.

I'll give this a shot this weekend.

KyleStilkey, 2024-04-10:
Sorry for the delay had some major storms in the north east here and took out power, but trying again seems that the servers are not responding, not sure if something has changed with the communication or if Tuya's servers are just under maintenance.

Here is the error, but I also do see that they pushed out a new EULA today that needs to be agreed on, not sure if this is any cause to the error either but again just providing any details I see.

E: An error has occurred
E: RuntimeError: Error: REMOTE_API_RUN_UNKNOW_FAILED: The server is busy, please try again later
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltctplugin\upk2esphome\work_schema.py", line 62, in run_impl

kuba2k2, 2024-04-10:
I am not sure why this is happening. The previous account apparently got blocked or something, this one works but only sometimes. Other times it will show this error or "user not exist" error. I am not really able to fix this in any way currently.

KyleStilkey, 2024-04-10:
> I am not sure why this is happening. The previous account apparently got blocked or something, this one works but only sometimes. Other times it will show this error or "user not exist" error. I am not really able to fix this in any way currently.

No problem at all, I gave it a few and it pulled the data needed.

[image]

This is error I get when pulling the Tuya data now, I'm guessing it just doesn't have the data points for these on Tuya's API or is this something related to the tool that needs to be looked into further? Thanks again for the help!

kuba2k2, 2024-04-10:
It is not an error. It just can't possibly convert every value type. You need to manually adjust your YAML.

KyleStilkey, 2024-04-10:
> It is not an error. It just can't possibly convert every value type. You need to manually adjust your YAML.

No problem at all, just wanted to check on this. Appreciate it, I will close out this report. Thanks for the help!