libretiny-eu/ltchiptool #28: Unknown bootloader CRC - 0x2739019F - please report this on GitHub issues!

schneemaier, 2024-04-13:
Hello, 

reporting this issue with a supposedly bk72xx clone chip labled as cc8000.

I was able to download the firmware from the chip but got the above warning for the bootloader crc

Result of `ltchiptool -v flash info bk72xx`:
```
D: Found Local data snapshot (v1.5.1) - /usr/local/lib/python3.11/dist-packages/ltchiptool
I: Available COM ports:
I: |-- ttyUSB0 - CP2102 USB to UART Bridge Controller - CP2102 USB to UART Bridge Controller - Silicon Labs (10C4/EA60)
I: |   |-- Selecting this port. To override, use -d/--device
I: |-- ttyS0 - ttyS0
I: Connecting to 'Beken 72xx' on /dev/ttyUSB0 @ 115200
D: Linking: stage 0
W: Unknown bootloader CRC - 0x2739019F - please report this on GitHub issues!
D: Reading 4k page at 0x011000 (0.00%)
D: BK72xx connected - protocol: BASIC_BEKEN, chip: None, bootloader: None, chip ID: None, boot version: None
D: Flash size - detecting...
D:  - Checking wraparound at 0x211000
D: Flash size detected - 0x200000
I: |-- Success! Chip info: Unknown
I: Reading chip info...
I: Chip: Unknown
I: +-----------------------+--------------+
I: | Name                  | Value        |
I: +-----------------------+--------------+
I: | Protocol Type         | BASIC_BEKEN  |
I: | Chip Type             | Unrecognized |
I: | Bootloader Type       | Unrecognized |
I: | Chip ID               | N/A          |
I: | Boot Version String   | N/A          |
I: |                       |              |
I: | Flash Size (detected) | 2 MiB        |
I: +-----------------------+--------------+
I: |-- Finished in 1.953 s
```

Thanks

kuba2k2, 2024-04-14:
Hi, we are aware of the clone chips. It seems they come with a variety of different bootloaders too.

Can you share the firmware you retrieved from the chip?

Does your device print anything on the UART logs? We would like to confirm the behaviour of these things.

Also, do note that neither ESPHome (LibreTiny) nor OpenBeken currently support the CC8000. Flashing the firmware will not make it boot, if it even succeeds at all.

schneemaier, 2024-04-14:
Thanks! I am aware that neither of the firmwares work on the device. 

Attached the cc8000 flash backup. The chip is installed in a bulb which works with the AiDot software. I don't have UART logs, but it general the UART logging is very limited on this firmware.  Here are some pictures of the modul: https://www.elektroda.com/rtvforum/viewtopic.php?p=21025914#21025914

[cc8000.zip](https://github.com/libretiny-eu/ltchiptool/files/14970771/cc8000.zip)

kuba2k2, 2024-04-14:
Just curious, how did you read the firmware from the device? I see OpenBeken's flasher tool doesn't work, and I've heard that bk7231tools/ltchiptool had some issues as well.

schneemaier, 2024-04-14:
> Just curious, how did you read the firmware from the device? I see OpenBeken's flasher tool doesn't work, and I've heard that bk7231tools/ltchiptool had some issues as well.

ltchiptool CLI worked for me. The GUI on Ubuntu had issues as it crashed during the download

```
ltchiptool flash read bk72xx ./cc8000.bin
I: Available COM ports:
I: |-- ttyUSB0 - CP2102 USB to UART Bridge Controller - CP2102 USB to UART Bridge Controller - Silicon Labs (10C4/EA60)
I: |   |-- Selecting this port. To override, use -d/--device
I: |-- ttyS0 - ttyS0
I: Connecting to 'Beken 72xx' on /dev/ttyUSB0 @ 115200
W: Unknown bootloader CRC - 0x2739019F - please report this on GitHub issues!
I: |-- Success! Chip info: Unknown
I: Reading Flash (2 MiB) to './cc8000.bin'
  [################################################################]  100%          
I: |-- Finished in 188.672 s
```

kuba2k2, 2024-04-14:
Great. Writing will not work unless you use the latest version of bk7231tools (development version, latest commit).

You can try booting ESPHome or your own code (like simple printf() in `void setup()`), maybe you'll get somewhere with this chip. I'm waiting for mine to arrive.

schneemaier, 2024-04-15:
> Great. Writing will not work unless you use the latest version of bk7231tools (development version, latest commit).
> 
> You can try booting ESPHome or your own code (like simple printf() in `void setup()`), maybe you'll get somewhere with this chip. I'm waiting for mine to arrive.

Thanks, i will not try to flash this board. as it is still usable with the bulb. I may order a separate single board with cc8000 to experiment :)

divadiow, 2024-04-21:
Chats about the HLK-B30:

https://discord.com/channels/967863521511608370/1229825856491491441

schneemaier, 2024-04-21:
> Chats about the HLK-B30:
> 
> https://discord.com/channels/967863521511608370/1229825856491491441

Discord link leads "NO TEXT CHANNELS" message

divadiow, 2024-04-21:
Oh. It's the dev board thread in the LibreTiny dumps channel on Discord