libretiny-eu/ltchiptool #29: Unknown bootloader CRC - 0xB7BFA0CE - please report this on GitHub issues

mchipser, 2024-04-26:
[image]

kuba2k2, 2024-04-26:
We will need more details. What kind of device is that? Is it a Tuya device? What bootloader version does it have?
We will also need a full firmware dump.

mchipser, 2024-04-26:
I will try and get that info.. [This is an Atomi string lights]https://timgray.blogspot.com/2021/12/hacking-atomi-smart-wifi-color-string.html(url), I think this is the right dump... 
[readResult_BK7231N_QIO_2024-25-4-23-09-14.zip](https://github.com/libretiny-eu/ltchiptool/files/15135343/readResult_BK7231N_QIO_2024-25-4-23-09-14.zip)

Cossid, 2024-04-26:
Looks like the 1.0.1 BK7231N bootloader, but the app is ESPHome, so not an original backup.

mchipser, 2024-04-26:
Does cloudcutter save a backup of the firmware somewhere?

Cossid, 2024-04-26:
No, it is not capable of doing firmware backups.  Part of the exploit process is overwriting factory data anyway.

mchipser, 2024-04-26:
I have another device, I will get it uploaded once I am able to dump it..

kuba2k2, 2024-04-27:
You might have flashed something incorrectly. If the dump has 1.0.1 bootloader, it should be recognized. If you've flashed something else and your device doesn't boot, it won't be recognized.
It's recommended to restore the full dump to fix the device.

J3r3Xy, 2025-02-09:
Hello! I had this exact message trying to read the flash of a CB2S module (BK7231N). The produced dump is just 1 byte. So i tried reading chip info, which failed with `E: ValueError: Read length 0x1000 is larger than flash memory size (0x1)`
So I tried using bk7231tools, which worked a little better:
```
$ bk7231tools chip_info -d /dev/ttyUSB0 
Unknown bootloader CRC - 0xB7BFA0CE - please report this on GitHub issues!
Reading 4k page at 0x011000 (0.00%)
BK72xx connected - protocol: FULL, chip: BK7231N, bootloader: None, chip ID: 0x7231c, boot version: None
Connected! Chip info: BK7231N / Flash ID: 00 00 00 / Flash size: 0x1 / Protocol: FULL
```
using read_flash with bk7231tools produced a 2MB binary filled with zeroes.
After a while spent switch OS, UART-USB converter, resoldering wires etc. I modified bk7231tools to allow reading past the 2MB and tried again, which (surprisingly to me) actually worked:
```
$ bk7231tools read_flash -d /dev/ttyUSB0 -s 0x200000 -l 0x200000 dump.bin -D
<- TX: BkLinkCheckCmnd()
-> RX (1): BkLinkCheckResp(value=0)
<- TX: BkCheckCrcCmnd(start=0x0, end=0x100)
-> RX (4): BkCheckCrcResp(crc32=0x1EBE6E45)
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
-> RX (1): BkLinkCheckResp(value=0)
<- TX: BkReadRegCmnd(address=0x800000)
-> RX (8): Response check OK
-> RX (8): BkReadRegResp(address=0x800000, value=0x7231C)
BK72xx connected - protocol: FULL, chip: BK7231N, bootloader: BK7231N_1_0_1, chip ID: 0x7231c, boot version: None
<- TX: BkFlashReg24ReadCmnd(cmd=0x9F)
-> RX (5): BkFlashReg24ReadResp(status=0, data0=0xEB, data1=0x60, data2=0x15)
Connected! Chip info: BK7231N / Flash ID: eb 60 15 / Flash size: 0x200000 / Protocol: FULL
Reading 2097152 bytes from 0x200000
Reading 4k page at 0x200000 (0.00%)
<- TX: BkFlashRead4KCmnd(start=0x400000)
-> RX (4101): Response check OK
-> RX (4101): BkFlashRead4KResp(status=0, start=0x400000, data=bytes(4096))
<- TX: BkCheckCrcCmnd(start=0x400000, end=0x400FFF)
-> RX (4): BkCheckCrcResp(crc32=0xA84F8983)
Reading failure @ 0x200000 (Chip CRC value 57B0767C does not match calculated CRC value 7F8F5109), retrying (attempt 0)
<- TX: BkFlashRead4KCmnd(start=0x400000)
-> RX (4101): Response check OK
-> RX (4101): BkFlashRead4KResp(status=0, start=0x400000, data=bytes(4096))
<- TX: BkCheckCrcCmnd(start=0x400000, end=0x400FFF)
-> RX (4): BkCheckCrcResp(crc32=0xA84F8983
Reading 4k page at 0x201000 (0.20%)
[...]
```
The generated dump 
[dump.bin.tar.gz](https://github.com/user-attachments/files/18724192/dump.bin.tar.gz) dissects correctly 
[dump_dissect.tar.gz](https://github.com/user-attachments/files/18724196/dump_dissect.tar.gz)
I retried reading chip info from ltchiptool, which this time worked:
```
+-----------------------+-------------------------------------+
| Name                  | Value                               |
+-----------------------+-------------------------------------+	
| Protocol Type         | FULL                                |
| Chip Type             | BK7231N                             |
| Bootloader Type       | BK7231N 1.0.1                       |
| Chip ID               | 0x7231c                             |
| Boot Version String   | N/A                                 |
|                       |                                     |
| MAC Address           | C0:F8:53:44:77:B0                   |
|                       |                                     |
| Flash ID              | EB 60 15                            |
| Flash Size (by ID)    | 2 MiB                               |
| Flash Size (detected) | 2 MiB                               |
|                       |                                     |
| Encryption Key        | 510fb093 a3cbeadc 5993a17e c7adeb03 |
+-----------------------+-------------------------------------+
```
I retried again after power cycling the module, and got again the error about read length larger than flash size (0x1), but forcing bk7231tools to read past 2MB still works. I can provide more info if needed!

J3r3Xy, 2025-02-11:
Yesterday I studied the code and learnt a couple things: first, flash addresses wrap around, so the successful read the other day was a coincidence, second, 0xB7BFA0CE is the CRC for 256 zero bytes.
So I did some more testing, turns out that the chip communicates correctly, but any command that tries to directly read flash returns zero. Writing the flash also does not work, though the chip answers normally to the write flash commands. Curiously, the command that asks the chip for the CRC for a chunk of flash data, appears to work correctly (I didn't thoroughly check, but forcing CRC checking in fact causes flash reading to fail with CRC errors, where the calculated CRC is always the CRC for a 4k page of zeroes, while the chip reports a different CRC for different pages).
At this point I had removed and resoldered all the wires, and cound't read anything at all from the chip.
I had read somewhere that some of these boards require CSN to be pulled LOW for flashing to work. I had flashed some other BK7231 and never needed that, but it was warth trying. It didn't work, but it gave me the idea to try to pull LOW each of the other pins while resetting the chip. When I pulled P24 LOW the program could read the chip info correctly. Tried again several times to be sure, and managed to read and write flash. Retried on identical CB2S with same result. 

TL;DR I had to pull P24 LOW while resetting the chip to be able to read or write flash.
Sharing in case someone else stumbles on this issue!

P.S. might be worth putting a note about this in the connection guide?

markrudling, 2025-02-13:
> TL;DR I had to pull P24 LOW while resetting the chip to be able to read or write flash. Sharing in case someone else stumbles on this issue!
> 

Thank you for taking the time to describe this - This solved the issue for me. The way it exhibits itself and how it differs from other very similar devices was very confusing.

kuba2k2, 2025-02-13:
Interesting - I'm not sure why would that work, this pin is not special in any way.

@J3r3Xy Nice find. It might be worth putting a note about this in the docs 😄

zastrixarundell, 2025-05-08:
> ```
> $ bk7231tools chip_info -d /dev/ttyUSB0
> Unknown bootloader CRC - 0xB7BFA0CE - please report this on GitHub issues!
> Reading 4k page at 0x011000 (0.00%)
> BK72xx connected - protocol: FULL, chip: [...]

Similar story. Although it is with an CB36. I was actually able to flash it with OpenBeken... kinda...

What worked was BK7231GUIFlashTool although *specifically* on windows. I've ran it in a container with mono installed and it didn't want to properly read or write to the flash.

Doing the P24 to LOW didn't help.

The weird issue is that I was NOT able to perform OTA's. OpenBeken OTA didn't work. When I installed ESPHome (and waited for it to boot for like 20 minutes), I was not able to perform OTA there too.

I'm stuck with using Windows 10 or setup a VM specifically for this singular device.

Changing flashers didn't help, CP2102 and CH340G behave the same regarding this.