libretiny-eu/ltchiptool #31: ltchiptool.gui.base.zc is deprecated ...

Another-Taipan, 2024-05-27:
When I launch v4.11.1 of Ltchiptool on Windows 10, I get this warning (in yellow text):

W: PyInstaller\loader\pyimod02_importers.py:419: UserWarning: ltchiptool.gui.base.zc is deprecated, migrate to ltchiptool.gui.mixin.zc

Is that important and what should I do to satisfy/fix the warning?

kuba2k2, 2024-05-27:
It's not important, it just means that one of the plugins hasn't been updated to the new Zeroconf module (probably UPK2ESPHome).

Another-Taipan, 2024-05-27:
OK, but when I attempt a "Read Flash" on a BK7231T with version 4.11.1, it almost completes and then throws this error:

E: An error has occurred
E: ValueError: No response received
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltchiptool\gui\work\flash.py", line 75, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 189, in _do_read
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 256, in flash_read_raw
E: |-- File "bk7231tools\serial\cmd_ll_chip.py", line 62, in read_flash_range_crc
E: |-- File "bk7231tools\serial\protocol.py", line 144, in command

The resulting dump file is 2,044 KB when it should be 2,048 KB.
I have repeated this process several times with exactly the same result.

However, if I do the same "Read Flash" with version 4.4.1, it completes without error and the dump file is the correct 2,048 KB.

andreas-bulling, 2024-11-24:
Same problem here but in my case the whole app crashes (Linux, installed with pip)