libretiny-eu/ltchiptool #33: Error writing ESPhome kickstart uf2 with ltchiptool v4.11.1

gurrier, 2024-07-14:
I'm attempting to flash with kickstart-bk7231n-2023-12-15.uf2
and get this reported error after it has flashed all the way to 100%
W: The current command timeout of 1.0 second(s) is too low for reading 905216 bytes CRC. Increasing to 3 second(s).

I don't see the Access Point advertised over wifi as expected while still connected to the flashing rig, so presume it has not worked.

There were no issues reading the existing firmware and generating YAML from that dumped firmware so the connection looks good.

gurrier, 2024-07-14:
The chip is in an Aldi (australia) 5 port powerboard with 2x USBC and 2 USBA.
Marked "CBU"

kuba2k2, 2024-07-14:
The error you got is just a warning message, it can be ignored safely.

However, if you're not getting the WiFi access point, the most likely cause is your flashing rig's power supply. Try adding a dedicated 3.3V power regulator, or disconnect the rig and run the device from its own power supply.

You can also try connecting TX2 to view UART logs.

gurrier, 2024-07-14:
Thanks for the response.
It appears you are correct.  The firmware was written.

Once I powered the device back up without the flashing rig, the access point did advertise albeit very intermittently.  That's likely an issue best addressed with libretiny though.

closed 2024-07-14 by gurrier