libretiny-eu/ltchiptool #34: Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))

I-hate-2FA, 2024-07-30:
[image]
chip may or may not be supported, new to none esp, cannot tell
[image]

kuba2k2, 2024-07-30:
Try running Get chip info.

It might be falsely detecting the chip type (T/N/etc) which could cause incorrect reading options.

Also, please post the full log of the flash operation (even better with enabled debug log level).

I-hate-2FA, 2024-07-31:
[image]
```
W: PyInstaller\loader\pyimod02_importers.py:419: UserWarning: ltchiptool.gui.base.zc is deprecated, migrate to ltchiptool.gui.mixin.zc

I: Loaded settings from C:\Users\Admin\AppData\Roaming\ltchiptool\gui.json
I: Found new device: COM11 - Silicon Labs CP210x USB to UART Bridge - Silicon Labs (10C4/EA60)
D: Log level changed
D: Started FlashThread
D: Starting READ_INFO operation; file = C:\Users\Admin\Downloads\ltchiptool_bk72xx_2024-07-31_03-11-32.bin, port = COM11 @ 115200
D: Connecting
D: Connecting
D: Connecting
D: Connecting
D: Reading 4k page at 0x000000 (0.00%)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 0)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 1)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 2)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 3)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 4)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 5)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 6)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 7)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 8)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 9)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 10)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 11)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 12)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 13)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 14)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 15)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 16)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 17)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 18)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 19)
W: Reading failure @ 0x0 (Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))), retrying (attempt 20)
E: An error has occurred
E: ValueError: Non-zero status code: BkFlashRead4KResp(status=6, start=0x0, data=bytes(1))
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltchiptool\gui\work\flash.py", line 69, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 109, in _link
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 102, in flash_connect
E: |-- File "bk7231tools\serial\linking.py", line 28, in connect
E: |-- File "bk7231tools\serial\linking.py", line 91, in detect_chip
E: |-- File "bk7231tools\serial\cmd_hl_flash.py", line 100, in flash_read_bytes
E: |-- File "bk7231tools\serial\cmd_hl_flash.py", line 85, in flash_read
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 105, in flash_read_4k
E: |-- File "bk7231tools\serial\protocol.py", line 196, in command
D: Stopped FlashThread
D: Started TerminalThread
D: Terminal opened

```

I-hate-2FA, 2024-07-31:
so is this chip supported by libretiny platform, or did i just wasted my money , i thought all 7231 is supported so i bought too much of them at once

kuba2k2, 2024-07-31:
It can be supported, don't give up yet.

Apparently this bootloader could be locking readout from 0x0.

Try using bk7231tools to read the flash chip contents:
https://github.com/tuya-cloudcutter/bk7231tools
```
bk7231tools read_flash -d COM11 -s 0x11000 -l 0x200000 dump.bin
```

I-hate-2FA, 2024-07-31:
```
C:\Users\Admin>bk7231tools read_flash -d COM11 -s 0x11000 -l 0x200000 dump.bin
Unknown bootloader CRC - 0xBEFD79D6 - please report this on GitHub issues!
Reading 4k page at 0x011000 (0.00%)
BK72xx connected - protocol: BASIC_BEKEN, chip: None, bootloader: None, chip ID: None, boot version: None
Flash size - detecting...
 - Checking wraparound at 0x211000
Flash size detected - 0x200000
Connected! Chip info: Unknown / Flash ID: None / Flash size: 0x200000 / Protocol: BASIC_BEKEN
Reading 0x200000 bytes at 0x11000 would go past the flash memory end
```

kuba2k2, 2024-07-31:
Right, so it seems it detects the flash size correctly.
Try with `-s 0x0`.

I-hate-2FA, 2024-07-31:
[dump.bin.txt](https://github.com/user-attachments/files/16439590/dump.bin.txt)
it seems to work, i used a brand new chip for this, cuz idk if delid have damaged it

kuba2k2, 2024-07-31:
Just curious, where did you find these BK7231S chips? We've never seen this particular model before, are these available to buy somewhere as bare chips?

I-hate-2FA, 2024-07-31:
> Just curious, where did you find these BK7231S chips? We've never seen this particular model before, are these available to buy somewhere as bare chips?

https://fcc.report/FCC-ID/2AR7VC-8133U/4225568.pdf
i bought 15 smart bulb for $4, it have this chip,isolated mains driver, pwm controlled RGBCCT

I-hate-2FA, 2024-07-31:
[image]

[image]

I-hate-2FA, 2024-07-31:
is there any chance for this to work

kuba2k2, 2024-07-31:
Yes. You can probably just flash ESPHome, provided that you use this in your YAML:
```yaml
esphome:
  name: my-device-name
  platformio_options:
    board_build.bkcrypt_coeffs: 00000000000000000000000000000000
    board_build.bkota.key: 0123456789ABCDEF0123456789ABCDEF
    board_build.bkota.iv: 0123456789ABCDEF
    board_flash.calibration: 0x1F3000+0x1000
```

Note that the bootloader on this device is the oldest we've seen, it looks very much different from all the standard bootloaders.
Because of this, I'm not sure if this device will support OTA as-is.

After compiling ESPHome, try flashing the `image_bk7231t_app.0x011000.rbl` file from the build directory to offset 0x11000, using the tool that works for you.

As long as you don't flash anywhere in between 0x0 and 0x11000, you should be fine.

I-hate-2FA, 2024-07-31:
what cpu should i select in esphome
is `board: generic-bk7231n-qfn32-tuya` correct

kuba2k2, 2024-07-31:
You should use `generic-bk7231t-qfn32-tuya`, not the N variant. T is the closest to your chip.

I-hate-2FA, 2024-07-31:
it cannot compile
[image]

kuba2k2, 2024-07-31:
Hm, maybe YAML is converting the coeffs to integer. Try wrapping it in quotes, like `board_build.bkcrypt_coeffs: "00000000000000000000000000000000"`.

I-hate-2FA, 2024-07-31:
what version do i download for cli tool

kuba2k2, 2024-07-31:
the Beken Application Image for direct flashing

I-hate-2FA, 2024-07-31:
didnt work still unfortunately
[image]
or maybe i messed up

I-hate-2FA, 2024-07-31:
it is not very easy to flash this because the cp2102 code not have auto reset, and i lost my ch340

kuba2k2, 2024-07-31:
Unfortunately I don't have a fix for that. It's not possible for me to fix such low-level issues without physically owning such a chip.

You could try flashing the same RBL file using other software, such as BkWriter. It should only be used as a last resort.

I-hate-2FA, 2024-07-31:
how can i send you one of these

kuba2k2, 2024-07-31:
If you don't live anywhere near Poland, I think it would be much more expensive than the entire device is worth :smile:

I-hate-2FA, 2024-07-31:
it is def more expensive then this is worth, but if you cant buy it there its the only option
it is only 1.5RMB per unit i bought from china

I-hate-2FA, 2024-07-31:
https://item.taobao.com/item.htm?id=749971518349

I-hate-2FA, 2024-07-31:
beken writer erase flash also fail

divadiow, 2024-07-31:
what version of bkwriter was it?

I-hate-2FA, 2024-07-31:
1.60

divadiow, 2024-07-31:
could try 1.7.5

https://images.tuyacn.com/smart/electronic/component/infomation/cfxjtyy7me4g.zip

I-hate-2FA, 2024-07-31:
same problem
[image]