libretiny-eu/ltchiptool #37: Unknown bootloader CRC - 0x0939FCDA - please report this on GitHub issues

dwmw2, 2024-08-11:
This is a Tongou TO-Q-SY1-JWT DIN rail power switch/meter: https://www.elektroda.com/rtvforum/topic3990985.html

It had already stopped working, so I got brave and opened it up. Appear to flash firmware just fine (ESPHome, then [OpenBK7231N_QIO_1.17.654.bin](https://github.com/openshwprojects/OpenBK7231T_App/releases/download/1.17.654/OpenBK7231N_QIO_1.17.654.bin)

```
|-- Unknown bootloader CRC - 0x0939FCDA - please report this on GitHub issues!
|   |-- Success! Chip info: BK7231N
|-- Writing '/home/dwmw/Downloads/OpenBK7231N_QIO_1.17.654.bin'
|   |-- Start offset: 0x11000 (auto-detected)
|   |-- Write length: 1.1 MiB (auto-detected)
|   |-- Skipped data: 68 KiB (auto-detected)
  [################################################################]  100%          |-- The current command timeout of 1.0 second(s) is too low for reading 1150976 bytes CRC. Increasing to 3 second(s).

|   |-- Finished in 61.297 s

```

Still doesn't work afterwards. The LEDs come on but there's no apparent WiFi activity, nothing on the serial port if I configure ESPHome to use UART1. But it was like that before with the Tuya firmware (although it was only the status LED that came on, IIRC, while now both do).

Will order a new one to sacrifice and see if that one works.

kuba2k2, 2024-08-11:
Your device most likely needs different encryption keys. OpenBeken currently doesn't have means to change encryption keys, LibreTiny/ESPHome does.

But in order to find the encryption keys for your device, you will need to post a stock firmware dump here.

dwmw2, 2024-08-11:
Thanks. What would the failure mode be in that case? It seemed to flash successfully but then just doesn't boot; is that what we'd expect?

New unit will be here tomorrow.

kuba2k2, 2024-08-11:
Yes, flashing a firmware with different keys will make it unable to boot.

Depending on whether you've flashed OpenBeken using their official tool (which overwrites the bootloader) you might be able to still retrieve the stock bootloader firmware dump. That could be enough to uncover the encryption keys.

dwmw2, 2024-08-11:
I only flashed with ltchiptool, so should a 'flash read' give us the bootloader?

kuba2k2, 2024-08-11:
Yes. With no parameters it should give a full 2 MiB firmware dump, which will most likely include the stock bootloader and your firmware flashing attempts.

dwmw2, 2024-08-11:
http://david.woodhou.se/frombk.bin

SHA256 f414981edc35e348524ae21a73295ac4522191f38d83f1bce5c0104cf4b391ca

kuba2k2, 2024-08-11:
This device uses default encryption keys - it's a Tuya device, they all use the same keys.

Flashing ESPHome should work with no issues, using the default config. Make sure you're checking the right UART port.

The bootloader might be corrupted - the binary differs from a known-good BK7231N bootloader by one byte. Try flashing just the bootloader from the OpenBeken's firmware image:
```
ltchiptool flash write -s 0x0 -l 0x11000 OpenBK7231N_QIO_1.17.654.bin
```

Then flash the firmware of choice without passing `-s` and `-l`.

dwmw2, 2024-08-11:
I had originally flashed with `esphome run...`.

Will try the above.

dwmw2, 2024-08-11:
```
Aug 11 13:52:41 lantiq.infradead.org dnsmasq-dhcp[24570]: DHCPACK(tag.2) 172.31.31.176 a0:92:08:23:d8:00 upk2esphome-bk7231n
```
Just flashing the bootloader worked. Thanks!

closed 2024-08-15 by dwmw2

divadiow, 2024-08-18:
> Depending on whether you've flashed OpenBeken using their official tool (which overwrites the bootloader)

this is no longer the case. This can be checked if bootloader overwrite is desired
[image]

divadiow, 2024-08-18:
> OpenBeken currently doesn't have means to change encryption keys, LibreTiny/ESPHome does.

It's a lot easier with LT/ESPhome but OpenBeken can tried by changing key in https://github.com/openshwprojects/OpenBK7231N/blob/master/platforms/bk7231n/bk7231n_os/build.sh