libretiny-eu/ltchiptool #40: "expecting ACK got b\x15 for block"

geedsen, 2024-09-24:
Following this installation description , I am flashing a Wifi Stick for a Solis inverter.
EVerything worked fine, up to the actuall flashing. Downloading the original firmware worked fine. But uploading the generated uf2 file produced these errors: 
[image]
[image]
Googling for this error, there are only a few results. Maybe a python version issue? 
I am using the WIndows Gui, 
[image]

panzerknacker20111, 2024-10-07:
I have the exact same issue. Using the same firmwarefile with the OEM Image Tool flashes just fine. But ltchiptool starts to transfer and ends with the shown errors... Any idea? I can provide needed Debug information on request if it helps the devs .

geedsen, 2024-10-07:
Actually I flashed two dataloggers, and both gave these errors. But the flash was actually OK, so probably you can ignore the errors.