libretiny-eu/ltchiptool #41: Chip connection guide for RTL8720CF

jeremysalwen, 2024-10-02:
I'm interested in dumping my Tuya WBR3 (RTL8720CF) chip from my smart fan.  I'm trying to follow the instructions, but it seems like there is no chip connection guide for this chip?

https://docs.libretiny.eu/docs/platform/SUMMARY/ 

Does anyone know where I can look for information on how I should connect my chip?

panzerknacker20111, 2024-10-07:
I successfully read back the flash of my RTL8720CF. If you can post an picture of your pcb i surely can help you with that. Most of the time they have already the RX TX GND and 3v3 exposed. But that depends on the type of pcb ;)

jeremysalwen, 2024-10-08:
Thank you @panzerknacker20111  I actually found that by running LTChiptool and selecting the chip, it output connection instructions: 

[image]

In combination with the images on the [Tuya Datasheet page](https://developer.tuya.com/en/docs/iot/wbr3-module-datasheet?id=K9dujs2k5nriy) I think this will be enough for me to connect it.
[image]

It seems that some of the connections are on the back of the WBR3, and so it is necessary to desolder the board to access them.  I won't be able to do that for a couple of weeks, but let me know if everything above sounds correct.