libretiny-eu/ltchiptool #42: Error CRC bootloader

Mit4el, 2024-10-29:
Hello, I use libretiny and ltchiptools in my firmware. some devices flash normally, but now I'm faced with such a problem. This is a relay on a BK7231N chip

P.S. I have already figured out the pins, I tried openbeken, it is flashing normally

[image]

[image]

[image]

[image]

Mit4el, 2024-10-29:
> Do you have a stock firmware dump of the device?
> You have most likely corrupted the bootloader somehow, so it can't boot again.
> It is also possible that your device came with different bootloader keys and doesn't boot because of that.
> In either case, having a stock firmware dump is necessary.

I saved the stock firmware, I'll try to restore it. Is the firmware address stored in the ufl file during the build? Can I see it, maybe it's stitching the wrong way?

kuba2k2, 2024-10-29:
No, there isn't anything like "firmware address".
Try restoring the entire firmware dump (it should be 2 MiB in size).
In ltchiptool, open the file for writing, disable "auto-detect parameters" and clear the start/skip/length fields.
Then flash the file.
The device should be fully functional with its stock features.

Is it a Tuya Smart device, or a different brand? (like SBER, eWeLink, etc.)