libretiny-eu/ltchiptool #43: Segmentation fault on Ubuntu 24

andreas-bulling, 2024-11-24:
I installed ltchiptool as described in the [docs](https://docs.libretiny.eu/docs/flashing/tools/ltchiptool/#gui-application). I also pip installed `upk2esphome` and `zeroconf`, all good.

Upon running `ltchiptool/bin/ltchiptool gui` and clicking on UPK2ESPHome, however, sometimes the app crashes right away with a segmentation fault. Or latest when I click on "Grab from ESPHome Kickstart".

The only error I see on the console is

`(ltchiptool:2656246): Gtk-CRITICAL **: 12:00:48.191: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar`

Version 4.11.2

Cossid, 2024-11-24:
I think there is currently some sort of threading bug, and if you have a lot of ESPHome devices on your network, it takes some time for the list of all devices to process, and if you try to do anything while that list is populated, it will hit that threading bug.  Try to wait for the list to stop expanding or number of devices to stop increasing before taking action.

andreas-bulling, 2024-11-24:
I tried that and, indeed, the crash happened way later. But it still happened as I wanted to type in the IP address of a ESPHome Kickstart Device...

supertin, 2024-12-15:
I guess I'm a +1 to this... I mean, I think. I just see "Segmentation fault (core dumped)" at various times. No errors logged that I can see.
Happens if I click on the Plugins tab. Happens randomly while downloading the flash. Happens when flashing... Basically makes the tool useless since I can't flash anything.