libretiny-eu/ltchiptool #44: Can I use ltchiptool/UPK2ESPHome with esphome-kickstart on ESP8266 devices?

felipecrs, 2024-11-28:
I just realized one of my Tuya devices is actually ESP8266-based.

I'm wondering if I can use UPK2ESPHome to help me generate the ESPHome YAML for my device.

kuba2k2, 2024-11-28:
No. ESP devices use a completely different format of the configuration data. Same goes for RTL8710B, where the format is similar to the one of ESP devices.

Currently only the Beken format is supported in upk2esphome.

felipecrs, 2024-11-28:
I see. Thank you!

closed 2024-11-28 by felipecrs

kuba2k2, 2024-11-28:
However, it could be possible to at least display UPK data in ltchiptool. That would show you the GPIOs, but you'd have to write the config manually anyway.

I don't know what format do ESP binaries use. If you drop a full firmware dump onto UPK2ESPHome in ltchiptool, it will try to read it. It won't generate any valid configuration, but it will show the raw data in `Source data` tab (if the storage encryption format is the same, that is).

felipecrs, 2024-11-28:
Ok, I will try to get the original firmware dump. I lost it for this device, but I think I have another unit.

Out of curiosity, this is what happens when I try with kickstart:

[image]

felipecrs, 2024-11-29:
Unfortunately it also doesn't work with the firmware dump:

[image]

I was able to obtain the dump using Tasmotize. I'm attaching here in case it's any useful.

[backup_20241129_180115.bin.txt](https://github.com/user-attachments/files/17963029/backup_20241129_180115.bin.txt)

After downloading, remove the .txt extension from the file name.

Cossid, 2024-11-29:
There doesn't appear to be any pin configuration present in that dump.

felipecrs, 2024-11-29:
Well, turns out my device actually uses a Tuya MCU. Would that explain it? The Tuya MCU pins are:

```yaml
uart:
  rx_pin: GPIO13
  tx_pin: GPIO15
```

Cossid, 2024-11-29:
Yes, TuyaMCU doesn't have device configuration, as those UART pins are programmed into the firmware.

felipecrs, 2024-12-03:
@Cossid, sorry, I know this is completely unrelated, but, do you happen to know if I can extract any useful information from the dump regarding MCU communication?

I have an odd issue when running ESPHome on this device, which I'd love to fix: https://github.com/esphome/issues/issues/6503

Cossid, 2024-12-03:
I'll respond in the ESPHome thread, as it make slightly more sense there.