libretiny-eu/ltchiptool #45: crc error on firmware

Mit4el, 2024-11-29:
hi, can you help with the error in the platformio project. On other generic-bk7231n-qfn32-tuya boards, the firmware works well. there is a problem with reading the crc, openbeken is flashing normally and working, so the board and programmer are working.

```
"C:\Users\DIeMONd\.platformio\penv\.libretiny\Scripts\python.exe" -m ltchiptool -r -i 1 -L "C:\Users\DIeMONd\.platformio\platforms\libretiny" flash write -d COM7 -b 921600 .pio\build\bk7231t\firmware.uf2
|-- Detected file type: UF2 - iotm_tiny 0.0.1
|-- Connecting to 'Beken 7231T' on COM7 @ 115200
|   |-- Success! Chip info: BK7231S_1.0.5
|-- Writing '.pio\build\bk7231t\firmware.uf2'
|   |-- iotm_tiny 0.0.1 @ 2024-11-29 09:07:10 -> generic-bk7231t-qfn32-tuya
################################################################|-- The current command timeout of 1.0 second(s) is too low for reading 716800 bytes CRC. Increasing to 2 second(s).
|-- ValueError: Chip CRC value 0 does not match calculated CRC value 26A6581F
|   |-- File "C:\Users\DIeMONd\.platformio\penv\.libretiny\Lib\site-packages\bk7231tools\serial\cmd_ll_chip.py", line 73, in check_crc
*** [upload] Error 1
```

Cut Platformio.ini
```
[env:bk7231t]
extra_scripts = 
        pre:tools/lt_fsbuild.py
        pre:tools/lt_fsflash.py
lib_compat_mode = off
lib_deps = 
        LT_WebSockets
https://github.com/Mit4el/ESPAsyncUDP#master
        ${common_env_data.lib_deps_external}
        ${env:bk7231t_fromitems.lib_deps}
lib_ignore = EspSoftwareSerial, HTTPUpdate, WebSockets
platform = https://github.com/Mit4el/libretiny#master
framework = arduino
board = generic-bk7231t-qfn32-tuya
custom_fw_name = iotm_tiny
custom_fw_version = 0.0.1
upload_speed = 921600
monitor_speed = 115200
build_flags = 
        -Dbk7231t="bk7231t"
        -DLT_LOGLEVEL=LT_LEVEL_DEBUG
        -DLT_DEBUG_ALL=1
        -DPROJECT_DATA_DIR="data_svelte"
        -DLT_USE_TIME=1
        -DDEBUG_ESP_PORT=Serial1
        -DNODEBUG_WEBSOCKETS=1
        -DLT_UART_DEFAULT_PORT=1
        -DUPLOAD_PORT=auto
build_src_filter = 
        +<*.cpp>
        +<classes/*.cpp>
        +<utils/*.cpp>
        +<modules/*.cpp>
        ${env:bk7231t_fromitems.build_src_filter}
```

kuba2k2, 2024-11-29:
Hi,
this is the third time we're seeing this issue lately, so something odd must be going on.

Are you able to flash the same UF2 file using the .EXE version of ltchiptool? (the GUI)
That version has old (known good) versions of all libraries, so this would let us determine if the issue is caused by an update of some dependency.

Also, a list of package versions present in your environment could help:

```
"C:\Users\DIeMONd\.platformio\penv\.libretiny\Scripts\python.exe" -V
"C:\Users\DIeMONd\.platformio\penv\.libretiny\Scripts\python.exe" -m pip list
"C:\Users\DIeMONd\.platformio\penv\.libretiny\Scripts\python.exe" -m ltchiptool -V
```

gDIeMONd, 2024-12-03:
"C:\Users\DIeMONd\.platformio\penv\.libretiny\Scripts\python.exe" -V
Python 3.12.3

"C:\Users\DIeMONd\.platformio\penv\.libretiny\Scripts\python.exe" -m pip list
Package            Version
------------------ ---------
bitstruct          8.19.0
bk7231tools        2.0.2
certifi            2024.8.30
charset-normalizer 3.4.0
click              8.1.7
colorama           0.4.6
hexdump            3.3
idna               3.10
importlib_metadata 8.5.0
ltchiptool         4.11.2
pip                23.2.1
prettytable        3.12.0
py-datastruct      1.1.0
pycryptodome       3.21.0
pyserial           3.5
requests           2.32.3
semantic-version   2.10.0
setuptools         65.5.0
urllib3            2.2.3
wcwidth            0.2.13
xmodem             0.4.7
zipp               3.21.0

C:\Users\DIeMONd\.platformio\penv\.libretiny\Scripts\python.exe" -m ltchiptool -V
ltchiptool v4.11.2

gui 
[image]

kuba2k2, 2024-12-03:
Can you enable verbose logging in ltchiptool GUI and post the full flashing log?

gDIeMONd, 2024-12-03:
I: Found new device: COM1 - Последовательный порт
I: Found new device: COM7 - USB-SERIAL CH340 - wch.cn (1A86/7523)
I: Connect UART1 of the BK7231 to the USB-TTL adapter:
I: 
I:     --------+        +--------------------
I:          PC |        | BK7231             
I:     --------+        +--------------------
I:          RX | ------ | TX1 (GPIO11 / P11) 
I:          TX | ------ | RX1 (GPIO10 / P10) 
I:             |        |                    
I:         GND | ------ | GND                
I:     --------+        +--------------------
I:  
I: Using a good, stable 3.3V power supply is crucial. Most flashing issues
I: are caused by either voltage drops during intensive flash operations,
I: or bad/loose wires.
I:  
I: The UART adapter's 3.3V power regulator is usually not enough. Instead,
I: a regulated bench power supply, or a linear 1117-type regulator is recommended.
I:  
I: To enter download mode, the chip has to be rebooted while the flashing program
I: is trying to establish communication.
I: In order to do that, you need to bridge CEN pin to GND with a wire.
V: Log level changed
V: OnUpdate(busy={'com7'})
V: OnUpdate(busy={'com7'})
V: OnUpdate(busy={'com7'})
V: OnUpdate(busy={'com7'})
V: OnUpdate(busy={'com7'})
V: OnUpdate(busy={'com7'})
D: Connecting
V: OnUpdate(busy=set())
V: OnUpdate(busy=set())
V: OnUpdate(busy={'com7'})
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
D: Connecting
V: OnUpdate(busy=set())
V: OnUpdate(busy=set())
V: OnUpdate(busy={'com7'})
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
V: <- TX: BkLinkCheckCmnd()
[... 1400 lines cut ...]
V: <- TX: BkFlashEraseBlockCmnd(erase_size=32, start=0x270000)
W: Erasing failure @ 0x70000 (No response received), retrying (attempt 3)
E: An error has occurred
E: ValueError: No response received
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltchiptool\gui\work\flash.py", line 73, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 126, in _do_write
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 314, in flash_write_uf2
E: |-- File "ltchiptool\util\streams.py", line 177, in update_from
E: |-- File "bk7231tools\serial\cmd_hl_flash.py", line 185, in program_flash
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 227, in flash_erase_block
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 190, in do_erase
E: |-- File "bk7231tools\serial\protocol.py", line 144, in command
V: Update: target=NoneType, port=COM7, family=<Family parent: children(2), name=beken-72xx, parent=None>
V: Update: target=ComboBox, port=COM7, family=<Family parent: children(2), name=beken-72xx, parent=None>
V: OnWorkStopped(<class 'ltchiptool.gui.work.flash.FlashThread'>)
V: OnUpdate(busy=set())
V: Deactivating page: <class 'ltchiptool.gui.panels.flash.FlashPanel'>
V: Activating page: <class 'ltctplugin.terminal.gui.TerminalPanel'>
V: OnUpdate(busy=set())
D: Stopped FlashThread
D: Started TerminalThread
D: Terminal opened
V: OnUpdate(busy=set())
V: OnUpdate(busy=set())

kuba2k2, 2024-12-04:
The log you posted shows a different error (`No response received`). That means the device was disconnected for one reason or another. If you want, try again to capture logs where the CRC error happens.

sredni, 2024-12-17:
I have similar problem, log output:
```
ltchiptool -vvv -r flash write -d /dev/cu.usbserial-017C078B  ~/Downloads/small-room.uf2
Found Local data snapshot (v1.7.0) - /Users/sredni/.local/pipx/venvs/ltchiptool/lib/python3.12/site-packages/ltchiptool
Detection: Detection(name='/Users/sredni/Downloads/small-room.uf2', size=2543616, type=<Type.VALID_UF2: 'UF2'>, file_type='UF2 - esphome 2024.8.3', offset=0, skip=0, length=2543616, family=<Family: BK7231T(0x675A40B0), name=beken-7231t, parent=beken-7231>, soc=None)
Detected file type: UF2 - esphome 2024.8.3
Connecting to 'Beken 7231T' on /dev/cu.usbserial-017C078B @ 115200
Linking: stage 0
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
<- TX: BkLinkCheckCmnd()
-> RX (1): BkLinkCheckResp(value=0)
<- TX: BkCheckCrcCmnd(start=0x0, end=0x100)
-> RX (4): BkCheckCrcResp(crc32=0x45AB3E47)
<- TX: BkBootVersionCmnd()
-> RX (16): BkBootVersionResp(version=b'BK7231S_1.0.5\x00\x00\x00')
BK72xx connected - protocol: BASIC_TUYA, chip: BK7231T, bootloader: BK7231S_1_0_5, chip ID: None, boot version: BK7231S_1.0.5
|-- Success! Chip info: BK7231S_1.0.5
Writing '/Users/sredni/Downloads/small-room.uf2'
|-- esphome 2024.8.3 @ 2024-12-17 11:05:42 -> wb2s
Writing (0x011000)  [----------------------------------------------------------------]    0%Erasing and writing at 0x11000 (0.00%)
<- TX: BkFlashEraseBlockCmnd(erase_size=32, start=0x211000)
-> RX (6): Response check OK
<- TX: BkFlashWrite4KCmnd(start=0x211000, data=bytes(4096))
-> RX (5): Response check OK
-> RX (5): BkFlashWrite4KResp(status=0, start=0x211000)
<- TX: BkCheckCrcCmnd(start=0x211000, end=0x212000)
-> RX (4): BkCheckCrcResp(crc32=0x9858B53A)
<- TX: BkLinkCheckCmnd()
-> RX (1): BkLinkCheckResp(value=0)
Erasing and writing at 0x12000 (0.52%)
<- TX: BkFlashEraseBlockCmnd(erase_size=32, start=0x212000)
-> RX (6): Response check OK
<- TX: BkFlashWrite4KCmnd(start=0x212000, data=bytes(4096))
-> RX (5): Response check OK
-> RX (5): BkFlashWrite4KResp(status=0, start=0x212000)
<- TX: BkCheckCrcCmnd(start=0x212000, end=0x213000)
-> RX (4): BkCheckCrcResp(crc32=0x9EECC985)
<- TX: BkLinkCheckCmnd()
-> RX (1): BkLinkCheckResp(value=0)
Writing (0x011000)  [----------------------------------------------------------------]    1%Erasing and writing at 0x13000 (1.04%)
<- TX: BkFlashEraseBlockCmnd(erase_size=32, start=0x213000)
-> RX (6): Response check OK
<- TX: BkFlashWrite4KCmnd(start=0x213000, data=bytes(4096))
-> RX (5): Response check OK
-> RX (5): BkFlashWrite4KResp(status=0, start=0x213000)
<- TX: BkCheckCrcCmnd(start=0x213000, end=0x214000)
-> RX (4): BkCheckCrcResp(crc32=0xEB8B0FB9)
<- TX: BkLinkCheckCmnd()
-> RX (1): BkLinkCheckResp(value=0)
[... 1896 lines cut ...]
-> RX (5): BkFlashWrite4KResp(status=0, start=0x2D1000)
<- TX: BkCheckCrcCmnd(start=0x2D1000, end=0x2D2000)
-> RX (4): BkCheckCrcResp(crc32=0xE3FB225A)
<- TX: BkLinkCheckCmnd()
-> RX (1): BkLinkCheckResp(value=0)
Verifying CRC
The current command timeout of 1.0 second(s) is too low for reading 790528 bytes CRC. Increasing to 2 second(s).
<- TX: BkCheckCrcCmnd(start=0x211000, end=0x2D2000)
-> RX (4): BkCheckCrcResp(crc32=0x2CE0F28D)
<- TX: BkLinkCheckCmnd()
-> RX (1): BkLinkCheckResp(value=0)
OK!
Writing (0x129F0A)  [###############################################################-]   99%  00:00:00Writing unaligned data...
<- TX: BkFlashEraseBlockCmnd(erase_size=32, start=0x329000)
-> RX (6): Response check OK
<- TX: BkFlashWriteCmnd(start=0x329F0A, [...]
-> RX (6): Response check OK
-> RX (6): BkFlashWriteResp(status=0, start=0x329F0A, written=102)
<- TX: BkCheckCrcCmnd(start=0x329F0A, end=0x329F70)
-> RX (4): BkCheckCrcResp(crc32=0xFFFFFFFF)
<- TX: BkLinkCheckCmnd()
-> RX (1): BkLinkCheckResp(value=0)
ValueError: Chip CRC value 0 does not match calculated CRC value 1DFBE182
|-- File "/Users/sredni/.local/pipx/venvs/ltchiptool/lib/python3.12/site-packages/bk7231tools/serial/cmd_ll_chip.py", line 73, in check_crc
```

prokoma, 2024-12-17:
It seems like ltchiptool is trying to write out of bounds of the flash (`start=0x329F0A`), could you please share the uf2 file?

Cossid, 2024-12-17:
Technically, all of its write are out of bounds by 0x200000 and wrap around as a work-around for protected areas, so 0x329F0A becomes 0x129F0A

It seems all bk7231t uf2 files have this issue, including the kickstart bin; https://github.com/libretiny-eu/esphome-kickstart/releases/download/v23.12.15/kickstart-bk7231t-2023-12-15.uf2

prokoma, 2024-12-17:
It would be helpful to read the flash back to determine whether something was actually written at 0x329F0A and the error is in the CRC command or if the write itself also failed.

Cossid, 2024-12-17:
I'll try to do so as time allows, will be sometime today.  Feel free to join the discord channel for discussion as well; https://discord.com/channels/967863521511608370/1146821147644657734

When this happens, we do know that the write of the app partition offset is actually successful, and the device does reboot to it.  There is a different BK7231T issue with bin files though (or at least used to be, but I don't remember how to verify/reproduce that at the moment).

kuba2k2, 2024-12-17:
The device indeed responds with a CRC value of 0xFFFFFFFF (or 0 if inversed). The logs are really helpful in this case.

Looking at the disassembly for BK7231S_1.0.5, there seems to be a problem with unaligned access to the flash. In particular, it reads chunks of 256 bytes into RAM, then calculates the CRC. It can't read properly if the addresses are not aligned.

```c
      crc_start = v1[1] | (v1[2] << 8) | (v1[3] << 16) | (v1[4] << 24);
      crc_end = v1[5] | (v1[6] << 8) | (v1[7] << 16) | (v1[8] << 24);
      crc_init_table();
      v6 = 0;
      v7 = crc_end - crc_start + 1;
      while ( v6 < v7 >> 8 )
      {
        flash_read(&bufa, crc_start, 256);
        crc_value = crc_calc(crc_value, &bufa, 0x100u);
        crc_start += 256;
        v6 = (v6 + 1) & 0xFFFEFFFF;
      }
      flash_protect_maybe(1);
      *v16 = crc_value;
      v16[2] = BYTE2(crc_value);
      v16[3] = HIBYTE(crc_value);
      cmd = uart_write_response(16, 8, v16);
```

Fixing this will require a change in bk7231tools, which also accounts for the unaligned data writing when checking the CRC.

Cossid, 2024-12-17:
Here is the before and after.  Before is a Tuya devboard stock firmware.  I wrote the same https://github.com/libretiny-eu/esphome-kickstart/releases/download/v23.12.15/kickstart-bk7231t-2023-12-15.uf2 kickstart uf2 file.

Full backup of device before uf2 write: [wb3s-pre-uf2.zip](https://github.com/user-attachments/files/18172502/wb3s-pre-uf2.zip)
log of writing uf2 file with CRC error: [kickstart_uf2_write_log.txt](https://github.com/user-attachments/files/18172500/kickstart_uf2_write_log.txt)
log of reading post-uf2 write: [kickstart_post_uf2_write_read_log.txt](https://github.com/user-attachments/files/18172499/kickstart_post_uf2_write_read_log.txt)
full backup of device post-uf2 write: [wb3s-post-uf2.zip](https://github.com/user-attachments/files/18172501/wb3s-post-uf2.zip)