libretiny-eu/ltchiptool #47: Unable to Install Tool

KyleStilkey, 2024-12-04:
When trying to install this tool I am given an error below.

```
E: An error has occurred
E: RuntimeError: Couldn't find embeddable package URL
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltchiptool\gui\work\install.py", line 31, in run_impl
E: |-- File "ltchiptool\util\ltim.py", line 93, in install
E: |-- File "ltchiptool\util\ltim.py", line 181, in _install_python_windows
```

I'm going to take a guess the python path has changed and is causing the install to fail.

kuba2k2, 2024-12-04:
The latest version 4.11.3 contains the fix, however I've been unable to build an EXE package yet.
If you want to install the tool, do this:
```
pip install --upgrade ltchiptool[gui]
python -m ltchiptool.util.ltim
```
This will install it to a directory called `ltchiptool` in portable version.

KyleStilkey, 2025-01-11:
I'm going to close this, I've gotten the 4.11.4 and it installes without any trouble.

closed 2025-01-11 by KyleStilkey