libretiny-eu/ltchiptool #51: Unable to get device schema from tuya cloud - Error: SING_VALIDATE_FALED

nefe2020, 2025-01-04:
Hi, I'm trying to convert this meter based on CB3S with TuyaMCU to ESPHome: https://www.amazon.it/dp/B0DKFYMHVZ
I successfully did a full dump of the flash memory but when loading into the UPK2ESPHome plugin to get the yaml configuration it says (correctly) that it needs to download the schema from tuya api, but when I press the download button I get the following error:
E: An error has occurred
E: RuntimeError: Error: SING_VALIDATE_FALED: SING_VALIDATE_FALED
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltctplugin\upk2esphome\work_schema.py", line 62, in run_impl

am I missing something?
Thanks

bavci1988, 2025-01-07:
I have the same issue with a eMylo TH16 Thermometer.
Not sure what that error means?

kuba2k2, 2025-01-08:
The error has been fixed, please try now. You don't need to upgrade any software to try.

nefe2020, 2025-01-08:
I confirm it's now working like a charm. Thanks a lot!

closed 2025-01-08 by nefe2020