libretiny-eu/ltchiptool #53: Unknown SCTRL_CHIP_ID - 0xfc011c

sosudaniel, 2025-01-11:
I am trying to flash a CB3S chip using the ltchiptool. Got the following error: W: Unknown SCTRL_CHIP_ID - 0xfc011c - please report this on GitHub issues!
Please let me know how I can help. The device is an Avatto whs20s.

sosudaniel, 2025-01-11:
W: Reading failure @ 0x1d0000 (Incomplete response read: 127 != 4101), retrying (attempt 0)
also the above warning appears

kuba2k2, 2025-01-11:
Check your wiring, it looks like you have some issues with either loose wires or weak power supply.

sosudaniel, 2025-01-11:
device is powered via its own cables, since my flashed cannot provide 3.3V. I would exclude this issue.

resolder all the wires, and the connections are good.

while testing found out that the CEN pin does not do anything. there is another PIN (RES) on the device that resets the device correctly.

sosudaniel, 2025-01-11:
then I shall postpone this until i will be able to remove the the tuyaMCU