libretiny-eu/ltchiptool #57: Can't copy text from 'Get Chip info' Window

Niaxor, 2025-02-03:
'Get Chip info' is critical for obtaining encryption keys from chips, but you can't copy them to the clipboard for use elsewhere as far as I can tell.

[image]

kuba2k2, 2025-02-03:
You should be able to use Ctrl+C to copy the entire window.

Also, you probably don't need to copy the encryption key, as it's usually the default one from Tuya (I can't tell because you censored it). But the fact that you have a 1.0.1 bootloader indicates likely that it's a Tuya chip and uses the default key. I can also see it ends with "3", which is likely "...093". I don't remember the entire key, but it starts with 5 and ends with 093.

Niaxor, 2025-02-04:
Ctrl+C to copy the entire window does not work - I double-checked this, as I'm aware you can do this for native message boxes on windows. I checked again after your comment.

Thank you for your additional comments on my encryption key - you are correct in that I did not end up needing it, but it seems a useful feature regardless 👍 

Feel free to close if you don't feel it is useful or necessary.