libretiny-eu/ltchiptool #58: Flashing BK7231 Tuya TH08 keeps crashing mid-way

paulandrewcrouch, 2025-02-08:
Any help appreciated.   I flashed my BK7231N chip with OpenBekin (it's a Tuya TH08 Temp and Humidity display).    I then tried to do an OTA update to ESP Home, something went wrong and nothing worked.   I've used Itchiptool to try and put ESP Kickstart, or an original binary back on to no avail.   I've erased all, re-created the RF partition, tried putting OpenBekin back on, and tried Itchiptool.   It just keeps failing at various degrees of the erase or flash process.   I have two, and both are doing the same.   It's driving me nuts.   I've used an external 3.3V Power Supply (presume no special grounding required)?   Any advice appreciated, so annoying given I once got OpenBekin on, and now can't seem to do anything!   Thank you.

kuba2k2, 2025-02-08:
The fact that you're getting random crashes would suggest a power supply issue.
Make sure you use a good power supply which can provide enough current. Also, check if your wires are connected properly.
By "special grounding" I assume you don't mean the fact that GND from the power supply has to be connected (it has to).

paulandrewcrouch, 2025-02-09:
> The fact that you're getting random crashes would suggest a power supply issue. Make sure you use a good power supply which can provide enough current. Also, check if your wires are connected [...]

Thanks for the quick reply, very powerful 3.3V external, by grounding I mean both the power supply -ve and the FT232 needs the ground connected together and connected to the chip?   I’ve tried different USB leads, have just ordered a new FT232 in case it was that.   I feel I’m so close, yet so far away!!

kuba2k2, 2025-02-09:
All parts of the setup (FT232, the chip, the power supply) must have their grounds connected together. Otherwise it simply won't work.

paulandrewcrouch, 2025-02-09:
> All parts of the setup (FT232, the chip, the power supply) must have their grounds connected together. Otherwise it simply won't work.

All connected, just tried again with a new FT board I ordered, cannot even connect now.   Could it be Windows drivers?   Totally perplexed.   I appreciate your help.   Thank you.

kuba2k2, 2025-02-09:
Post what kind of error are you getting while flashing. Make sure to enable verbose logging in ltchiptool before that.

kuba2k2, 2025-02-11:
It would be best if you flashed a full factory firmware dump. Doing that, remember to uncheck "auto-detect" in ltchiptool and set start and skip to 0x0, length to 0x200000.

However, the bootloader is not active during the flashing process, so corrupting the bootloader won't make a difference during flashing.

paulandrewcrouch, 2025-02-11:
`W: Writing 4k failure @ 0x0 (No response received), retrying (attempt 0)
W: Erasing failure @ 0x0 (No response received), retrying (attempt 0)
W: Erasing failure @ 0x0 (No response received), retrying (attempt 1)
W: Erasing failure @ 0x0 (No response received), retrying (attempt 2)
W: Erasing failure @ 0x0 (No response received), retrying (attempt 3)
E: An error has occurred
E: ValueError: No response received
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltchiptool\gui\work\flash.py", line 73, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 161, in _do_write
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 286, in flash_write_raw
E: |-- File "ltchiptool\util\streams.py", line 177, in update_from
E: |-- File "bk7231tools\serial\cmd_hl_flash.py", line 182, in program_flash
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 185, in flash_write_4k
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 244, in flash_erase_block
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 203, in do_erase
E: |-- File "bk7231tools\serial\protocol.py", line 144, in command
E: Caused by ValueError: No response received
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 173, in flash_write_4k
E: |-- File "bk7231tools\serial\protocol.py", line 144, in command
E: An error has occurred
E: SerialException: ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltctplugin\terminal\work.py", line 38, in run_impl
E: |-- File "serial\serialwin32.py", line 259, in in_waiting
I: Device unplugged: COM4 - USB Serial Port - FTDI (0403/6001)
I: Found new device: COM4 - USB Serial Port - FTDI (0403/6001)
W: Unknown bootloader CRC - 0xB397B8F6 - please report this on GitHub issues!
W: Writing 4k failure @ 0x4f000 (No response received), retrying (attempt 0)
W: Erasing failure @ 0x4f000 (No response received), retrying (attempt 0)
W: Erasing failure @ 0x4f000 (No response received), retrying (attempt 1)
W: Erasing failure @ 0x4f000 (No response received), retrying (attempt 2)
W: Erasing failure @ 0x4f000 (No response received), retrying (attempt 3)
E: An error has occurred
E: ValueError: No response received
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltchiptool\gui\work\flash.py", line 73, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 161, in _do_write
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 286, in flash_write_raw
E: |-- File "ltchiptool\util\streams.py", line 177, in update_from
E: |-- File "bk7231tools\serial\cmd_hl_flash.py", line 182, in program_flash
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 185, in flash_write_4k
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 244, in flash_erase_block
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 203, in do_erase
E: |-- File "bk7231tools\serial\protocol.py", line 144, in command
E: Caused by ValueError: No response received
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 173, in flash_write_4k
E: |-- File "bk7231tools\serial\protocol.py", line 144, in command
W: Writing 4k failure @ 0x0 (No response received), retrying (attempt 0)
W: Erasing failure @ 0x0 (No response received), retrying (attempt 0)
W: Erasing failure @ 0x0 (No response received), retrying (attempt 1)
W: Erasing failure @ 0x0 (No response received), retrying (attempt 2)
W: Erasing failure @ 0x0 (No response received), retrying (attempt 3)
E: An error has occurred
E: ValueError: No response received
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltchiptool\gui\work\flash.py", line 73, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 161, in _do_write
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 286, in flash_write_raw
E: |-- File "ltchiptool\util\streams.py", line 177, in update_from
E: |-- File "bk7231tools\serial\cmd_hl_flash.py", line 182, in program_flash
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 185, in flash_write_4k
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 244, in flash_erase_block
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 203, in do_erase
E: |-- File "bk7231tools\serial\protocol.py", line 144, in command
E: Caused by ValueError: No response received
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 173, in flash_write_4k
E: |-- File "bk7231tools\serial\protocol.py", line 144, in command
E: An error has occurred
E: SerialException: ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
[... 483 lines cut ...]
E: |-- File "ltchiptool\gui\work\flash.py", line 73, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 161, in _do_write
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 286, in flash_write_raw
E: |-- File "ltchiptool\util\streams.py", line 177, in update_from
E: |-- File "bk7231tools\serial\cmd_hl_flash.py", line 182, in program_flash
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 185, in flash_write_4k
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 244, in flash_erase_block
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 203, in do_erase
E: |-- File "bk7231tools\serial\protocol.py", line 144, in command
E: Caused by ValueError: No response received
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 173, in flash_write_4k
E: |-- File "bk7231tools\serial\protocol.py", line 144, in command
[... cut ...] beken-72xx, parent=None>
V: Update: target=ComboBox, port=COM4, family=<Family parent: children(2), name=beken-72xx, parent=None>
V: OnWorkStopped(<class 'ltchiptool.gui.work.flash.FlashThread'>)
V: OnUpdate(busy=set())
V: Deactivating page: <class 'ltchiptool.gui.panels.flash.FlashPanel'>
V: Activating page: <class 'ltctplugin.terminal.gui.TerminalPanel'>
V: OnUpdate(busy=set())
D: Stopped FlashThread
D: Started TerminalThread
D: Terminal opened
V: OnUpdate(busy=set())
V: OnUpdate(busy=set())
`

kuba2k2, 2025-02-11:
This is the most common error, and it usually means that there is something wrong with the way you supply power to the chip. Either the power supply is too weak/unstable, or your wires are not connected properly.

If you soldered the wires, check the soldering and see if it's okay. If you didn't... well then that's most likely the problem 🙂 

I can't really help any further, because I don't know your exact setup. If you can, maybe posting a photo of the components and connections would help somewhat.

paulandrewcrouch, 2025-02-11:
> This is the most common error, and it usually means that there is something wrong with the way you supply power to the chip. Either the power supply is too weak/unstable, or your wires are not [...]

Ok, in a way it's good to know it's common, and yes, I soldered, so I'm going to re solder, I presume I can just ignore the bootloader issue then?