libretiny-eu/ltchiptool #61: BW15-kit board, can't enter download mode due to RTS/DTR

agirilovich, 2025-04-13:
At first glance BW15-kit board looks a a very convinient device to play with libretiny. Everything is on board, many LEDs, the switch for Serail ports etc.
Unfortunately, ltchiptool can't flash it. As any other tool from either recommended or not-recommended tool from the list.

After some experiments I realised, that I still can upload firmware if USB port is used for power supply only,  external USB->RS232 adapter is  connected directly to RXD0, TXD0 pins and a default ceremeny of entering in download mode is used (hold PROG, click RST, release PROG).

So the problem is around embedded USB->RS232 serial adapter (CH340 chip).
According to datasheet, on this board, pin RTS of the CH340 chip is connected to CHIP_EN and pin DTR is controlling state of IO_0. 

I don't not specifically how pins RTS/DTR are managed in CH340, but looks like first chunk of data from  ltchiptool  just reset the board and turn it out of download mode.

A bit later I found out, that [AmebaZII_PGTool](https://github.com/Ameba-AIoT/ameba-rtos-z2/tree/main/tools/AmebaZ2/Image_Tool) works seamlessly with that board and switch ony flash mode in a fully automatic way. But I didn't how it is implemented.

bdraco, 2026-01-13:
Well not a good board to test with because OTA is a stub

cores/realtek-ambz2/base/api/lt_ota.c

bdraco, 2026-01-13:
https://github.com/libretiny-eu/libretiny/pull/302

pulleralarm, 2026-01-15:
I've added to following hack at the end of the __init__ method in ambz2tool.py to make flashing my BW15 board via USB work. In addition sleep has to be imported from the time module at the beginning of the file.

        print("Doing init Reset magic")
        self.s.dtr = True
        self.s.rts = True
        sleep(0.1)
        self.s.rts = False
        sleep(0.1)
        self.s.dtr = False

Setting DTR to True pull PA00 high, RTS then issues a reset via CEN. After 0.1 seconds, the chip should be in flash mode and PA00 can be lowered again to allow further boots.

agirilovich, 2026-09-29:
I think we should close the issue as fallen by the wayside.

In addition to all the problems with AmebaZ2 chips,  BW15-kit board is not produced anymore and out of stock.
Whatever solution we develop, it won't be well-tested, merged to the main branch and demanded ever.

For rare owners, the solution with direct connection to L_RX/L_TX pins is noted here.

closed 2026-09-29 by agirilovich