libretiny-eu/ltchiptool #62: Unknown bootloader CRC - 0x6BEB0924  (for a BK7238)

Cthanatos, 2025-04-17:
This is for a string of [Brizlabs Smart Fairy Lights](http://www.amzn.com/dp/B0DPHF2T8P). I'd be happy to donate the cost for you to order one yourself for testing.

It has "CBU" on the PCB, and the chip info  comes up with this:

[image]

I saw in another thread about pulling just the bootloader, so here it is from BK7231Flasher

[readResult_BK7238_QIO_BK7238-bootloader.zip](https://github.com/user-attachments/files/19786133/readResult_BK7238_QIO_BK7238-bootloader.zip)

Here's the full flash read from BK7231Flasher

[readResult_BK7238_QIO_full.zip](https://github.com/user-attachments/files/19786136/readResult_BK7238_QIO_full.zip)

and here's the Rom (I think I got it, let me know if I need to try again)

[ltchiptool_bk72xx_2025-04-16_16-06-08_rom.zip](https://github.com/user-attachments/files/19786150/ltchiptool_bk72xx_2025-04-16_16-06-08_rom.zip)

Not sure if this helps but here's the output from the UART2 bus right after I connect ground to CEN. Let me know if (and maybe how) to get whatever else you need.

```
vHeapInit-start addr:0x412f20, size:184544
[Flash]id:0x852015
[Flash]init over
sctrl_sta_ps_init
SDK Rev: 3.0.65 
[THD]app:[tcb]414200 [stack]4131f8-4141f8:4096:5
[THD]extended_app:[tcb]414a70 [stack]414268-414a68:2048:4
[THD]idle:[tcb]414ee0 [stack]414ad8-414ed8:1024:0
[THD]timer_thd:[tcb]415c68 [stack]415060-415c60:3072:2
OSK Rev: F-3.0.40 
cset:0 0 0 0
[FUNC]rwnxl_init
chip id=7238 device id=22068000
IP Rev: W4-3.0.65-P0
txdesc flush
[FUNC]intc_init
[FUNC]calibration_main
device_id=0x22068000
calibration_main over
NO TXPWR_TAB_TAB found in flash
Load default txpwr for b:0xe3e10
Load default txpwr for g:0xe3e46
fit n20 table with dist:4
Load default txpwr for ble:0xe3e1e
calibrate low value:[477]
calibrate high value:[8f0]
temp in flash is:255
[THD]temp_detct:[tcb]416680 [stack]415e78-416678:2048:3
lpf_i & q in flash is:74, 76
xtal in flash is:23
xtal_cali:23
--init_xtal = 23
[FUNC]ps_init
int watchdog enabled, period=10000
task watchdog enabled, period=60000
[FUNC]func_init_extended OVER!!!

start_type:0
[THD]kmsgbk:[tcb]417748 [stack]416740-417740:4096:6
[THD]init_thread:[tcb]417f88 [stack]4177b0-417f80:2000:5
Initializing TCP/IP stack
tcp_port:55452
[THD]tcp/ip:[tcb]4189b8 [stack]4181b0-4189b0:2048:7
[THD]wpas_thread:[tcb]419e80 [stack]418e78-419e78:4096:4
bk_wlan_app_init finished
[THD]core_thread:[tcb]41ae38 [stack]41a630-41ae30:2048:7
[THD]rf_arbitrate:[tcb]41b7a0 [stack]41af98-41b798:2048:8
rf_thread_init ok
[THD]ble:[tcb]41c8b8 [stack]41b8b0-41c8b0:4096:5
[... 176 lines cut ...]
wpa rx E SCAN_RESULTS
Scan completed in 1.120000 seconds
wpa_get_scan_rst:7
rlink_light_sys_app_init
rino_light_prod_init
ret:0, test_mode:2
Prod test ssid chang to test2!
Got ap count:88***66********* 7
wpa S: SCANNING -> DISCONNECTED
que type: 18, free memory 100576
Got ap count: 7
free memory 100192
que type: 5, free memory 100192
WRN|2021-01-01 00:00:03|rlink_ble.c|ble_send_packet_by_ntf(606): ble peer disconnect
ble send, ret=-1
tuya_hal_adc_value_get: timeout!++++++++++last_adc = 3039+++++++value:0
--write status reg:0,2--
--write status reg:7c,2--
--write status reg:0,2--
--write status reg:7c,2--
bk_save_env
adc_value=3041, adc_value_max=3755, adc_value_min=341, RGBWMic=47, gMicSens=330, g_max_connt=3000, SlaveMode=0x80
adc_value=3315, adc_value_max=3752, adc_value_min=2488, RGBWMic=47, gMicSens=330, g_max_connt=3000, SlaveMode=0x80
tuya_hal_adc_value_get: timeout!++++++++++last_adc = 3124+++++++adc_value=3124, adc_value_max=3750, adc_value_min=2516, RGBWMic=47, gMicSens=330, g_max_connt=3000, SlaveMode=0x80
```

Cthanatos, 2025-04-21:
Not sure if this is helpful or not, but here's some documentation from Beken about the [BK7238 hardware and CRC partitions](https://docs.bekencorp.com/sdk_3.0.x/bk7238/build/en/latest/developer-guide/partition_specification/index.html).

Ivkaiv, 2025-04-21:
I have the same chip and the same issue

Cthanatos, 2025-04-22:
I'm learning as I go, but I'm looking for useful info. I was able to get the UART CLI to work with the UART 2 going to a terminal window in ltchiptool, and UART 1 going through Putty. the "partition" command gave me this:

[image]

Here's a list of the commands I get using tab-completion (pressing tab with nothing else typed) 

[image]

No idea if any of those will give helpful info so I'm just tinkering and keeping an eye open.

kuba2k2, 2025-04-22:
The "unknown bootloader" is just a warning message, there are too many bootloader versions to list them all in the program.

It should work properly nevertheless - are you experiencing issues with flashing/dumping via ltchiptool?

Cthanatos, 2025-04-22:
I appear to be able to dump the flash. I'd like to move it to ESPHome, but when I try to analyze the firmware dump, I get this error. 

[image]

I've dropped a different firmware with BK7231Flasher that seemed to work, but it obviously wipes all the configuration data so I don't know what I need to have everything work the way I want it to from ESPHome.

kuba2k2, 2025-04-22:
This device is not connected to Tuya Cloud, it's a custom solution created by the manufacturer. UPK2ESPHome can't generate configuration for such devices.

That being said, BK7238 is not currently supported by ESPHome. You could try compiling ESPHome for BK7231N, but that probably wouldn't "just" work.

In any case, it's not an issue with ltchiptool, but a missing platform support in LibreTiny.

Cthanatos, 2025-04-22:
Good to know, that's what I was afraid of. Looks like I'll need to order a different string of lights that I know works with the tuya cloud so I can get the configs. Thank you!