libretiny-eu/ltchiptool #63: invalid literal for int() with base 16: 'DW'

Jerther, 2025-05-07:
Hi!

I'm trying to dump the firmware of a RTL8710CF chip. I can tell it is properly connected to the serial port as I can read activity.

```
ltchiptool flash info ambz2
```

I get this error:

```
I: Connecting to 'Realtek AmebaZ2' on /dev/ttyUSB0 @ 115200
E: ValueError: invalid literal for int() with base 16: 'DW'
E: |-- File "/home/jtheriault/.local/lib/python3.13/site-packages/ltchiptool/soc/ambz2/util/ambz2tool.py", line 187, in dump_words
```

The line the program is trying to parse is this:

```
DW 40000038 4
```

Any idea what's wrong?

kuba2k2, 2025-05-07:
This could mean that your wiring is unstable, and there are some missing bytes, and the readout gets desynchronized.
It could also mean that your chip has a different ROM version that is doing something differently, not how the program would expect.

Make sure to enable debug logging `-vvv`. Also for the `info` command, you can use dump serial activity (`-S` if I'm not mistaken) to dump all bytes passing through, hopefully letting to narrow down the possible causes of this issue.

(you can use -S for all commands, but it will spit out a lot of information if you're dumping the flash)

Jerther, 2025-05-07:
@kuba2k2 

Here are the results:

```
$ ltchiptool -vvv flash info ambz2

*snip!*

I: Connecting to 'Realtek AmebaZ2' on /dev/ttyUSB0 @ 115200
D: Linking: stage 0
V: push_timeout(0.1)
V: pop_timeout()
V: push_timeout(0.048)
E: ValueError: invalid literal for int() with base 16: 'DW'
E: |-- File "/home/jtheriault/.local/lib/python3.13/site-packages/ltchiptool/soc/ambz2/util/ambz2tool.py", line 187, in dump_words

```

```
$ ltchiptool -vvv -s flash info ambz2

*snip!*

I: Connecting to 'Realtek AmebaZ2' on /dev/ttyUSB0 @ 115200
D: Linking: stage 0
S: <- TX: 'Rtk8710C'
V: push_timeout(0.1)
S: -> RX: 'Rtk8710C'
V: pop_timeout()
S: <- TX: 'ping'
S: -> RX: 'wifi link err:00000400'
S: -> RX: 'ping'
S: -> RX: 'dissconn reason code: 0'
S: -> RX: 'scan stage, scan timeout'
S: <- TX: 'ping'
S: -> RX: 'ping'
V: push_timeout(0.048)
S: <- TX: 'DW 40000038 4'
S: -> RX: 'DW 40000038 4'
E: ValueError: invalid literal for int() with base 16: 'DW'
E: |-- File "/home/jtheriault/.local/lib/python3.13/site-packages/ltchiptool/soc/ambz2/util/ambz2tool.py", line 187, in dump_words

```

Is the MCU echoing?

Jerther, 2025-05-07:
I think it is. And all the unrelated messages is just WIFI debug data being sent to the serial port by the MCU.

Here's an example from start:

```
== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0

== Boot Loader ==
Dec  9 2020:20:15:00

Boot Loader <==

== RAM Start ==
Build @ 19:55:33, Oct 11 2022

$8710c>interface 0 is initialized
interface 1 is initialized

#######################################
        TP-LINK IoT Platform HS103(US)
        Software version: 1.0.8 Build 221011 Rel.195515
        Hardware version: 5.0
#######################################

Initializing WIFI ...
WIFI initialized

WIFI Mode No Need To Change: STA -->STA
[Driver]: set ssid [DOMOTIQUE] 

auto reconnect ...

[Driver]: set ssid [DOMOTIQUE] 
wifi link err:00000400
dissconn reason code: 0
scan stage, scan timeout

auto reconnect ...

[Driver]: set ssid [DOMOTIQUE] 
wifi link err:00000400
dissconn reason code: 0
scan stage, scan timeout

```

kuba2k2, 2025-05-07:
Ah, I see. You've not entered the download mode properly. The chip is still running the actual firmware, and apparently there is a command line enabled, which echoes back any character sent to the chip.

The real download mode would indeed respond `ping` to a `ping` command, so ltchiptool assumes that linking was successful. That wasn't the case, so trying the `DW` command makes the CLI simply echo it back.

Make sure you've entered the download mode, by powering up the device while pulling pin A0 high. On the serial port it should then mention that "download mode" has been entered and the firmware should not be running normally.

Jerther, 2025-05-07:
Oh I see!!

Then may I suggest the following change in this help message:

```diff
I: Connect UART2 of the Realtek chip to the USB-TTL adapter:
I: 
I:     --------+        +---------------------
I:          PC |        | RTL8720C            
I:     --------+        +---------------------
I:          RX | ------ | TX2 (Log_TX / PA16) 
I:          TX | ------ | RX2 (Log_RX / PA15) 
I:             |        |                     
I:         GND | ------ | GND                 
I:     --------+        +---------------------
I:  
I: Using a good, stable 3.3V power supply is crucial. Most flashing issues
I: are caused by either voltage drops during intensive flash operations,
I: or bad/loose wires.
I:  
I: The UART adapter's 3.3V power regulator is usually not enough. Instead,
I: a regulated bench power supply, or a linear 1117-type regulator is recommended.
I:  
- I: In order to flash the chip, you need to enable download mode.
+ I: In order to flash (read or write) the chip, you need to enable download mode.
I: This is similar to ESP8266/ESP32, but the strapping pin (GPIO 0 / PA00)
I: has to be pulled *to 3.3V*, not GND.
I:  
I: Additionally, make sure that pin PA13 (RX0) is NOT pulled to GND.
I:  
I: Do this, in order:
I:  - connect PA00 to 3.3V
I:  - apply power to the device OR shortly connect CEN to GND
I:  - start the flashing process
```

kuba2k2, 2025-05-07:
Yeah, "flashing" usually means either writing ("downloading") or reading ("uploading?") 😄 

Feel free to submit a PR if you're up to it. Otherwise, we'll keep this issue open and label it `documentation` so that I can fix it sometime in the future.

Jerther, 2025-05-07:
Success!!! 😀 

```
I: |-- Success! Chip info: RTL8720CF
I: Reading chip info...
I: Chip: RTL8720CF
I: +---------------------+-------------------+
I: | Name                | Value             |
I: +---------------------+-------------------+
I: | Chip VID            | 5                 |
I: | Chip Version        | 3                 |
I: | ROM Version         | v3.0              |
I: |                     |                   |
I: | Chip Type           | RTL87x0CF         |
I: | MAC Address (Wi-Fi) | FF:FF:FF:FF:FF:FF |
I: | MAC Address (BT)    | FF:FF:FF:FF:FF:FF |
I: | Boot Debugging      | Disabled          |
I: | Secure Boot         | Disabled          |
I: |                     |                   |
I: | Flash Type          | RTL8720CF         |
I: | Flash Mode          | SINGLE            |
I: +---------------------+-------------------+
I: |-- Finished in 4.953 s

```

closed 2025-05-07 by Jerther