libretiny-eu/ltchiptool #67: flash LN882x error

wiltet, 2025-07-19:
If I try to flash a LN-02 board with compiled .uf2 from ESPHome Builder I do get following errors:

1. run ltchiptool-v4.12.1.exe:
     FileNotFoundError: [WinError 3] system cant find path: \\AppData\\Local\\Temp\\-MEI51642\\itchiptool\\soc...\ramcode.bin

2. run (normal and as administrator) itchiptool-v4.12.1 installed program:
    PermissionError: Errno13 Permission denied: \\AppData\Local\\Temp\tmp....._

here is the log from itchiptool-v4.12.1 (installed):

[log.txt](https://github.com/user-attachments/files/21328127/log.txt)

I have newest Python 3.13.5 for Win installed on Win10 with newest updates.

Maybe I'm  doing something wrong, but can't find the cause of this error.

lamauny, 2025-07-22:
Can you add these options : verbose  (-v) and dump serial (-s) and post your logs again ?

wiltet, 2025-07-22:
thanks!
here are two full!? logs

[new1.txt](https://github.com/user-attachments/files/21376122/new1.txt)
[new1(2).txt](https://github.com/user-attachments/files/21376369/new1.2.txt)

Today I tryed to find faults on the UART cabelagem and soldered new short cables on two devices, but it made no difference.
BekenFlasher works without problems, but only with a baud rate of 115200.

kuba2k2, 2025-07-23:
If it only works at baudrate of 115200, that indicates a problem with your UART setup. LN882 works at a much higher baudrate, so if you have problems at anything above 115200 it will break. BK7231 chips should normally also work properly at 460800 or even 921600.

lamauny, 2025-07-23:
If you have access to the source files (it seems to), can you test replacing 
```
        tmp_file = NamedTemporaryFile()
        with open(tmp_file.name, "wb") as f:
```
with
```
        with NamedTemporaryFile() as f:
```
in ltchiptool\soc\ln882h\util\ln882htool.py", line 211

wiltet, 2025-07-25:
thanks, but it is not working yet.

[fix2 log.txt](https://github.com/user-attachments/files/21425745/fix2.log.txt)

I will check as well on a linux system to proof if the error is not on hardware side.

wiltet, 2025-07-25:
yes, this was it!

Flash Successful.

thanks a lot for your help!

-------------------------

May I ask a other thing. Sorry for offtopic, but I think it is easy for you to answer my question.
Meanwhile I have a Linux Mint installation for eperimenting and tried to install itchptool, but I struggle allready at first step installing wxPython. When I run: pip install -U \  -f https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-24.04 \  wxPython
it interrupts and I do get the information to do the install in some other way or install with pipx into a virtual environment. (sorry, do not have full text because I'm on Windows atm)
What do I need to do with this? Is pipx the right way?

lamauny, 2025-07-25:
Thanks for testing, I'll update the file ;-)

Concerning wxPython, I know its installation can be a pain but I don't remember what was the good way sorry...